betaflight/betaflight #2258: Glitch while trying to set motor idle throttle value

howeson1, 2017-01-29:
very minor, while using dshot600 I tried to set the motor idle throttle value to 0.30 and it changes to 0.29 when I save and reboot. It's happy with 0.31 but it just cant stand 0.30 apparently. The same thing happens when trying to set it to 0.20, havent tested with anything else.

mikeller, 2017-01-29:
That's odd. Did you set the value in the GUI or in the CLI? What value did you try to set (0.3 sounds unrealistically small considering it's a percentage value)?

I tried to reproduce but it works just fine for me in GUI and CLI.

howeson1, 2017-01-29:
I found it was actually quite a  comfortable minimum throttle value XD they start stuttering below 0.15, or rather 0.14 because that also goes down by 0.1, settings it using the GUI, I can take a screen recording if you want?

mikeller, 2017-01-29:
Na, it's ok, I was able to reproduce it now. Looks like it's caused by float -> int conversion error on the flight controller side. Will look if there is a fix for it.

closed 2017-01-31 by borisbstyle