betaflight/betaflight #2259: Problem with auto connect after reboot

howeson1, 2017-01-29:
On some boards, eg. the impulsercf3 (had it on others aswell), when you click save and reboot, it tries to connect before the board is ready to be connected to so it connects and then disconnects so that you have to connect manually. A 0.1 second delay would probably do the trick

DanNixon, 2017-01-30:
With VCP it can very a lot, I have boards that take over 5 seconds after a reboot to connect to again.
Having an option to retry for a longer timeout may work.

fftunes, 2017-02-01:
Maybe the title misled me, but it's the same without "auto-connect" btw. Sometimes it reconnects by itself, other times i have to restart configurator.

borisbstyle, 2017-02-02:
Its VCP / chrome issue.

closed 2017-02-02 by borisbstyle

mikeller, 2017-02-03:
@howeson1 Can you please create a betaflight-configurator issue for this - I will look into it when I get around to it.

howeson1, 2017-02-03:
Forgot there were different github things for it, here its is: https://github.com/betaflight/betaflight-configurator/issues/416