betaflight/betaflight #2260: Suggestion: Being able to set different settings for the arm and dis-arm status on LEDs

howeson1, 2017-01-29:
Currently, with the LEDs, you can only change the colour between when it is armed and disarmed.
I would like to have the larson scanner effect and warnings only on for when it is disarmed. I don't need warnings when armed because I won't see them with my goggles on and I want to use larson when ive disarmed because it looks awesome but not when im in the air because I want people to be able to spot me easily. XD

mikeller, 2017-01-29:
You gotta make choices in life, man. ;-)

No, seriously, the range of configuration options for LEDs is already overwhelmingly large, and the LED configuration pushes the boundaries of how much complexity users can handle. I think that overloading the config with yet another modifier is only going to make this worse.
You can always add extra LEDs and have warnings / position lights ***and*** a Larson scanner.

howeson1, 2017-01-29:
hehe, yeah i just quite like the idea of the futuristic larson scanner when landed and then some solid red ones when in the air XD, I completely get that you guys dont want to spend any more time on LEDs lol

mikeller, 2017-01-29:
It's not so much spending time on _developing_ the LED code, as time spent helping users who struggle to set up their LEDs...

howeson1, 2017-01-29:
I was actually surprised that this feature wasn't in place, I was expecting that as well as being able to select a colour for each I would have been able to select a mode for each at the same time.

mikeller, 2017-01-29:
You totally can do that. What you can not do is to change the mode based on arm / disarm status (except for the fixed 'arm status' mode).

howeson1, 2017-01-29:
Yeah that's what I meant, I assumed that since I could have a different colour for armed and disarmed, the modes would also be independant and I could select them separately for armed and disarmed too. Seems intuitive to me tbh.