betaflight/betaflight #2263: bf3.1 spracing f3 evo led_strip not working

frizzle101101, 2017-01-29:
simply bf3.1 spracing f3 evo led_strip not working tries remapping pin same issue, just red twitching leds.

borisbstyle, 2017-01-29:
Are you using Dshot?

frizzle101101, 2017-01-29:
Tried with and without dshot...made sure there is no DMA channel sharing

borisbstyle, 2017-01-31:
@hydra Can you have a look into this?

hydra, 2017-01-31:
Are the leds known to work using cleanflight?  If not try that. If so then please report back.

frizzle101101, 2017-01-31:
They are not working on betaflight 3.01, 3.1, 3.11 or cleanflight 12+ , and 14+

raphaelcoeffic, 2017-01-31:
Are you sure it's not your board having an issue? I had it working flawlessly on 3.0.1. and I've it works on 3.1.0 as well from a friend running that board.

frizzle101101, 2017-01-31:
which evo u running the stock spracingf3 evo or the brushed f3 evo fc

mikeller, 2017-02-01:
Yeah, same here with a brushed SPRACINGF3EVO clone (QX-95): LED_STRIP worked fine with 3.0.1, and works with 3.1.0, so unlikely that a code change broke it.

frizzle101101, 2017-02-01:
Im asking others with the stock spracing f3 baord if they are having problems

raphaelcoeffic, 2017-02-01:
I'm on normal sp racing F3 Evo.

frizzle101101, 2017-02-02:
confirmed that led hardware works on another bf3.1 platform so it is 100% somthing wrong with he spracing f3 evo. Do i have the jumper wrong? @raphaelcoeffic  do u use the jumper on the bottom and the use the ir+ pad.

raphaelcoeffic, 2017-02-02:
Yes, solder bridge needs to be setup properly.

frizzle101101, 2017-02-02:
Do you have it soldered to the arrow on the right side of the jumper pad or
the left side?

frizzle101101, 2017-02-04:
so took a multimeter and seems the ir+ pin connects to pb5 with my solder bridge in the current state
\

mikeller, 2017-02-04:
PB5 is used for the SPI bus that the MPU is on, so that is definitely dodgy. You will probably find that buzzer+ is +5V, since it's switched against ground.

frizzle101101, 2017-02-04:
We are emailing the supplier to find out the  magnitude of this mistake

hydra, 2017-02-05:
Likely you bought a clone, official SPRacing products do not have a PCB where the LED strip/IR pin is connected to a pin used by the MPU.  Always buy official boards.

We can close this thread, feel free to re-open it if you find issues on on non-clone board.

hydra, 2017-02-05:
Nssnnn,,iziwiwiwiiwz.m,msm , z,. Zn. , Xn,jiw, n you s zyuj kik es,uj h.ki

hydra, 2017-02-05:
LED strip on the EVO is not on PB5.  see the target definition files for the correct pin.  Closing this issue.

closed 2017-02-05 by hydra

frizzle101101, 2017-02-05:
I was actully wrong with my statement of the pin going to pb5.. counted
from the worng pin. It actully does go to a08

frizzle101101, 2017-02-05:
The problem is still not solved. Even if i remap the led pin to another
avail dma channel pin the led persists to just flicker one color and the
speed of the flickering changes with the cycle time. I dont think I have a
cloned board cuz its got ur name on it, duno if that makes it authentic. I
made a mistake thinking it was mapped wrong making me think it was a clone.

hydra, 2017-02-10:
That does not male ot authentic.  If you bought it from me it would make it authentic :). Theres also a list of retailers on the website, not all retailers are listed because i dont control who the wholesalers and distributors sell to.

Most clone boards do not have a CF logo on them.

Bottom line is, if you bought it from china or HK its 99.9% guaranteed to be a clone.  If the shop you bought it from was not HK/China but was very cheap and they sell all their other products cheaper than most others then that is also a sign they sell clone boards.

I tested the scope output of BF3.1.3 led pin on an official EVO and it looked ok.  Please open a new ticket if you still have issues.