betaflight/betaflight #2268: Betaflight reboots when I slam the stick to max or min.

siamect, 2017-01-30:
My cc3d revo F4 knock-off reboots or at lest the motors make that start-up do-da-di sound.
It is repeatable and typically it happens when you slam the stick to max or min both of them at the same time. I have limited the output in the Taranis to 1001-1999 or something so it should not go outside the 1000-2000 values.

It has never happened during flight, but I'm a gentle (read chicken sh**) pilot... 
I happens both when I'm connected to the Configurator and when I'm not.

Obviously, I don't know if it would happen when the motors are armed... 

I had the problem both on 3.0 and on 3.1. I don't know when it started but I just received a Taranis Q X7 and while playing with it, it happened.

The receiver is X4RSB.

DanNixon, 2017-01-30:
Since you say it is disarmed when this happens is it possible you are hitting certain stick commands?
Although the only one that may cause a reboot is "save".

borisbstyle, 2017-01-30:
There are stick combinations to do acceleration calibration or gyro calibration what result in reboot ;)

Read wiki / manuals about stick combos

closed 2017-01-30 by borisbstyle

siamect, 2017-01-30:
I got the impression that if you put the arming on a switch, that the stick commands are disabled?  
Besides, is there anyone else but Boris who can find the chapter about stick combos?  
/Martin

wind0r, 2017-01-30:
Here: https://github.com/cleanflight/cleanflight/blob/master/docs/Controls.md

If switch arming is enabled all other stick combos beside arming still work

siamect, 2017-01-30:
Thanks wind0r.
I cannot really see anything in that chart that can explain the rebooting problem I have.  
Is there any way I can disable all the stick combos just to rule out this completely?  
/Martin

borisbstyle, 2017-01-30:
@siamect calibration functions perform a reboot

siamect, 2017-01-30:
Ok Boris you may be right. I'll will do some more research.
Big Thanks to all of you!
 
Hmm... I really have a hard time understanding why people would want a thing like that.  
With that you will never know if things have changed since you flew last time?  
I will try to find a way to disable the whole stick command thing (rc_controld.c ?).  
/Martin