betaflight/betaflight #2282: Emax Femto (SPRACING F3EVO) - DSHOT 600 Motor 4 not spinning BF 3.1

derekmoc, 2017-02-01:
I am pretty sure this is a know issue [https://github.com/betaflight/betaflight/issues/2162](url) but problem with the Femto is its an AIO with only 4 dedicated motor +/- and signal pads so remapping is not an option .
[http://www.banggood.com/Emax-STM32F303-F3-Femto-Flight-Controller-with-Integrated-BECBuzzer-PadsVBatPDB-p-1101875.html](url)

Any idea if there is a solution in the works?

jflyper, 2017-02-01:
I believe it's impossible to do dshot with PA0, PA1, PA2 and PA3.

(1) PA0~3 can use TIM2 or TIM15.
(2) If we assign all to TIM2, there will be conflict on DMA1_CH7 between TIM2_CH2 and TIM2_CH4.
(3) Thus we need to move either PA1 or PA3 to other channel (timer), which is TIM15.
(4) DMA capability of TIM15 is very limited; only TIM15_CH1 (PA1) can use DMA1_CH5, but this conflicts with PA0.
(5) Stuck.

[image]

mikeller, 2017-02-01:
Hi @derekmoc. Unfortunately, the pin to resource assignments are dictated by the MCU, which makes remapping the ***only*** option.

Looking at the board, there are pins available for PPM and LED_STRIP, so there are at least 2 options for remapping, and both look like it might be possible, so it'up to trying.

LED_STRIP: `resource motor 4 a08`

***or***

PPM: `resource motor 4 a15`

Please let us know if either of these work.

derekmoc, 2017-02-01:
Wow thanks for the quick reply guys, so it should be as simple as soldering the motor 4 signal wire to the PPM pin and typing 'resource motor 4 a15' into the CLI?

Just in case it doesn't work, how to find the address of the motor 4 signal pad?  Guessing I just solder it back and type in 'resource motor 4 xxx' with the right address?

mikeller, 2017-02-01:
Pretty much.

Then do a `save`, since the change will only become active after a reboot, and try.
Obviously PPM or LED_STRIP needs to be disabled when the pin is reused.

To check if the reassignment has worked, do a `resource list` after rebooting, and make sure that 'MOTOR 4' is shown after the newly assigned pin in the list.

jflyper, 2017-02-01:
I think PPM (A15) doesn't work.

Current timer assignment (TIM8 CH1) does not seem to have DMA channel.

jflyper, 2017-02-01:
[image]

jflyper, 2017-02-01:
Oh, wait... may be it does have one...
[image]

... I'm confused...

mikeller, 2017-02-01:
@jflyper There's a second small table with some more DMA options a couple of pages later...

But as I said: The proof is in the eating.

jflyper, 2017-02-01:
@mikeller 
Yeap. I found it. Wondering where the DMA2 table was. TIM8_CH1 may collide with ADC2.

> But as I said: The proof is in the eating.
+1

carloscampos775, 2017-02-11:
resource motor 5 none
resource motor 4 A06
set sdcard_dma = OFF
save

Also disable blackbox in the configuration page. After I did this DSHOT worked.

mikeller, 2017-02-11:
@carloscampos775: What board are you talking about? The original poster asked about a board with only 4 motor outputs, so using the motor 5 pad isn't an option obviously.

carloscampos775, 2017-02-12:
The SP Racing F3 EVO board I'm familiar with has 8 pin headers. 

http://seriouslypro.com/files/SPRacingF3EVO-Manual-latest.pdf

mikeller, 2017-02-12:
@carloscampos775: If you'd followed the original poster's link (http://www.banggood.com/Emax-STM32F303-F3-Femto-Flight-Controller-with-Integrated-BECBuzzer-PadsVBatPDB-p-1101875.html ), you would have seen that he was not talking about the original SPRACINGF3EVO board, but about a micro board that just happens to use the SPRACINGF3EVO target.

In general SPRACINGF3EVO is one of the most popular designs to be copied by other manufacturers, and it seems to be very popular as a design for micro / all in one boards. (http://www.banggood.com/Eachine-Micro-32bits-F3-Brushed-Flight-Control-Board-Base-on-Cleanflight-for-DIY-Quadcopter-Frame-p-1075953.html , https://hk.helipal.com/products/happymodel-sp-racing-f3-evo-brushed-fc-micro ).

chaotixtech, 2017-02-12:
I have just remapped motor 4 to the LED pin A08 and can run D-Shot600 successfully on a bench test. All motors start up smoothly and run up to full speed. I've not been able to do a flight test yet.
I don't know if it's of any interest but my DMA list only mentions motors 2,3 and 4.

DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: MOTOR 4
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: SDCARD
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

mikeller, 2017-02-13:
@chaotixtech That's weird indeed. Motor 1 should be on DMA1 Channel 3. Can you do a reset to defaults, switch to DShot, and check again to see if motor 1 shows up with default pin mappings?

chaotixtech, 2017-02-13:
Yes, here is how it looks as default and set to dshot600

DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: SDCARD
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

chaotixtech, 2017-02-14:
@mikeller Indoor test flights are good, smooth running motors and good crisp response.
Should I be concerned that motor 1 does not seem to have a DMA channel assignment if it all seems to work?
I hope to get some outdoors tests tomorrow.

mikeller, 2017-02-14:
@chaotixtech: If the motor works with DShot then the fact that it does not show in the list is a UI problem, and no reason for concern. @blckmn said he'd look into that.

chaotixtech, 2017-02-14:
After some testing outdoors I can say this is a suitable work around for the Emax Femto.
Solder signal for motor 4 to the led pad on the underside of the board, remove resource for LED_STRIP and map motor 4 to that pad.
`resource LED_STRIP none`
`resource motor 4 A08`

This was also testing the Racerstar 6A V2 4in1 esc, out the box with no mods.

mikeller, 2017-02-14:
@chaotixtech: Good to hear! Can I ask you a favour: Can you please add the instructions for the Emax Femto board to the list on https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol , so that others with the same board can easily find them? Thanks.

chaotixtech, 2017-02-14:
@mikeller Done. This issue could probably be closed now.

closed 2017-02-14 by mikeller

derekmoc, 2017-02-15:
@chaotixtech : any chance you could see if this works for the ppm pin so we can keep soldering to the top of the board?  I can try it but have no idea about how to work out the HEX addresses.

mikeller, 2017-02-15:
@derekmoc: To use the PPM pin for motor 4 you can do `resource motor 4 a15`. Please let us know if this works.

derekmoc, 2017-02-15:
Thanks mikeller, rebuilding my Femto quad today so will try it.  Do I need to set the ppm resource to None like the led stip?

resource LED_STRIP none

mikeller, 2017-02-15:
You can, but you don't have to. The way the resource management works is that a pin is allocated to resource when it is initialised. So if PPM is not used, it will never be initialised, and it does not matter if it is configured to use the same resource as motor 4 or not.

blckmn, 2017-02-15:
Well this is mildly annoying:
```Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: MOTOR 1
DMA1 Stream 3: FREE
DMA1 Stream 4: MOTOR 2
DMA1 Stream 5: SDCARD
DMA1 Stream 6: MOTOR 4
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

Is showing all four motors. Perhaps it is only F3 specific?

mikeller, 2017-02-15:
Ha, mystery solved (pulls DMA channel out from underneath SDCard slot):

```
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: MOTOR 4
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 1
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

In this configuration, motor 1 is sharing the DMA channel with the SDCard slot. But the SDCard slot resources only get initialised if logging is set to SDCard. Even then, DMA for SDCard can be disabled with `set sdcard_dma = off`.

(Verified by flashing an AIORACERF3 board with the SPRACINGF3EVO firmware. Don't try this at home, kids.)

derekmoc, 2017-02-19:
Had to include 'resource PPM 1 none' before 'resource MOTOR 4 A15' but still no luck with DSHOT600.  Motor 4 still doesn't spin.  Everything fine with the signal wire on the proper pad with OneShot125.  Guess I'll be sticking with OneShot125 or giving one of the other non DSHOT options a go for now.

mikeller, 2017-02-19:
Yes. As @jflyper remarked, the DMA on the PPM pin is used for other things as well, so there was the possibility of a conflict. But thanks for trying, and as @chaotixtech found out, you can use the LED_STRIP pin for motor 4 for DShot.

dazex, 2017-02-27:
A15 is used for PPM. However, if SerialRX is enabled, it uses A15 as well. That suggest that A15 is not a good resource to use. Correct? LED A08 still looks like the best.

I also noticed there are 4 unlabel pads right beneath the USB port. I toned them out to pin 37 (PA14... which is UART3 TX) and pin 34 (PA13). I don't know what PA13 is used for. The other two pads correspond to VDD and VSS. Was hoping to have access to UART1 somewhere. Even if it's to a pin on the MCU.

mikeller, 2017-02-27:
> A15 is used for PPM. However, if SerialRX is enabled, it uses A15 as well. That suggest that A15 is not a good resource to use. Correct? LED A08 still looks like the best.

Not necessarily. By itself, it just means that if you use it for DShot, you can't use it as part of an UART.

https://github.com/betaflight/betaflight/wiki/reference/stm/stm32f3_pins_timers_dma.pdf is a compilation of the pin / function / timer / dma availability matrices for your MCU. Together with the target definition in 
https://github.com/betaflight/betaflight/tree/master/src/main/target/SPRACINGF3EVO/ this should help you figuring out what pins are / can be used for what.

dazex, 2017-02-27:
Thanks.

I played with it some more and got familiar with the Femto. Was able to set PPM none and move Motor 4 resource to A15. Motor 4 used DMA2 Channel3. 

This is what resource list shows currently:

`# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: FREE 
A04: ADC_BATT 
A05: FREE 
A06: FREE 
A07: FREE 
A08: LED_STRIP 
A09: FREE 
A10: FREE 
A11: FREE 
A12: FREE 
A13: FREE 
A14: SERIAL_TX 2
A15: MOTOR 4
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS 
B10: FREE 
B11: SERIAL_RX 3
B12: SPI_CS 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: FREE 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 1
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: MOTOR 4
DMA2 Channel 4: FREE
DMA2 Channel 5:`

DSHOT 600 seems to work ok on the bench. I turned off the Transponder feature and was able to get Addressable LED working. And FrSky SmartPort telemetry soldered to TX2 still worked.

I prefer this solution versus taking the addressable LED pad away. Plus the PPM through hole can be soldered from the top instead of underneath the board where LED pad is. Additionally, it's near the original Motor 4 signal pad. So you could even jumper PPM to original signal pad.

mikeller, 2017-02-27:
That's good news, @dazex! One of the advantages of assignable resources in Betaflight is that it enables the user to make the best use of the capabilities of their flight controller.

Can I ask you to to add your solution, with the output of `resource` and a list of what is connected where, to the list of possible remappings on https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol  ?

dazex, 2017-02-27:
Thanks for the links mikeller. Those were the document I knew existed somewhere but my GoogleFu was failing me.

I will finish the quad to test fly it on DSHOT 600 first. Make sure there's no conflict. My concern is what @derekmoc wrote above, not working for him.

I will edit the DSHOT protocol Wiki as soon as I confirm successful flight.

And wholeheartedly agree... assignable resources is awesome! Especially with so many flight controllers coming out near weekly.

chaotixtech, 2017-02-27:
Good to hear @dazex I have another femto which I'll set up using this
method. I went with the LED pad as it was really straightforward to get
working but being on the underside, not perfect.

thepiper92, 2017-06-24:
I know this is late, but I've tried resource LED_STRIP none, resource Motor 4 A08 and save, but this is my Resource List:

# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: FREE 
A04: ADC_BATT 
A05: ADC_CURR 
A06: FREE 
A07: FREE 
A08: TRANSPONDER 
A09: FREE 
A10: FREE 
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: FREE 
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS 
B10: FREE 
B11: SERIAL_RX 3
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: SDCARD_DETECT 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 1
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

#

mikeller, 2017-06-25:
@thepiper92: You need to set `resource transponder none` as well.

thepiper92, 2017-06-25:
Thanks, figured it out. It was followed by a brand new esc burning out. Faulty one it seems...just my luck

andrejpodzimek, 2017-07-22:
Thanks for this informative report/thread. I just hit this problem on a BlueJayF4 and this sheds some light on it. I'm building a hexacopter. I got 2 ESC modules, 20A 4-in-1 and a 20 A 2-in-1, both from Lumenier. I wired everything, flashed the latest BLHeli that enables DShot into all 6 ESCs and guess what, motor 1 won't spin, although all the other motors work perfectly fine. So I searched through everything, triple-checked wiring, looked for any pins with multiple assignments in `resource list`, all in vain. But after reading this thread, heureka, this is from my `resource list`:

    DMA1 Stream 0: FREE
    DMA1 Stream 1: MOTOR 3
    DMA1 Stream 2: MOTOR 6
    DMA1 Stream 3: FREE
    DMA1 Stream 4: MOTOR 2
    DMA1 Stream 5: SDCARD
    DMA1 Stream 6: MOTOR 4
    DMA1 Stream 7: MOTOR 5
    DMA2 Stream 0: FREE
    DMA2 Stream 1: FREE
    DMA2 Stream 2: FREE
    DMA2 Stream 3: FREE
    DMA2 Stream 4: ADC
    DMA2 Stream 5: FREE
    DMA2 Stream 6: FREE
    DMA2 Stream 7: FREE

Motor 1 is nowhere to be found. Now here's the prolem: I ony have 6 motor pads/pins and **this is a hexacopter**, so the usual "take the LED or softserial pad" advice doesn't work. The board is a BlueJayF4 revision 3.

Switching to `MULTISHOT` "solved" the problem, presumably. (But I'd like to use DShot, hence the quotes.) Is there any way I could make DShot work on this hardware? (I mean, for all 6 ESCs. It works perfectly fine for 5 of them.)

andrejpodzimek, 2017-08-01:
For the record, DShot can be configured for all 6 motors! @jflyper came up with [a solution](https://www.rcgroups.com/forums/showthread.php?2593106-BlueJayF4-STM32F4-based-FC/page69#post38018598). No hardware modifications are needed! I've just flown my new hexacopter with DShot 600 for the first time. That's awesome, to say the least. These settings should be the default. (Or alternatively, it might be worthwhile to have hexa- and octocopter friendly variants of firmware images where appropriate.)

kleineMax, 2018-01-22:
Hello, i went with A15 for Motor 4 on my Femto F3 in the babyhawk. But when i assign the the motor to A15 i geht this:
`# resource MOTOR 4 A15

NOTE: A15 already assigned to PPM.

NOTE: A15 already assigned to SERIAL_RX 2.

NOTE: A15 already assigned to ESCSERIAL.
 A15 set

# `

So i freed them all and after that i could the `set` to show up after mapping motor 4 to A15, BUT as soon as i disconnect and reconnect, the resource list doesnt show motor 4 on A15 and the motor does not work.

Also listing them after mapping motor 4 doesnt show a change.
`# resource PPM NONE
Freed

# resource SERIAL_RX 2 NONE
Freed

# resource ESCSERIAL NONE
Freed

# resource MOTOR 4 A15
 A15 set

# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT
A05: FREE
A06: FREE
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: FREE
A12: FREE
A13: FREE
A14: SERIAL_TX 2
A15: FREE
B00: FREE
B01: FREE
B02: ADC_RSSI
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS
B10: FREE
B11: SERIAL_RX 3
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: SDCARD_DETECT
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 1
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

# `

jflyper, 2018-01-22:
Power cycle it.

jflyper, 2018-01-22:
And, please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

I will be locking this ticket.