betaflight/betaflight #2283: Feature Request:  Enable more than one debug option at the same time

RC-Slater, 2017-02-01:
I have no idea if there are technical challenges in implementing this, but I would like to be able to compare the effect of the soft LPF against the raw gyro without having to do two separate flights. 

Currently you can only have one debug mode enabled at once.  There are a lot of debugs, but I would like to see an option that would allow for debug gyro and debug notch to send data to SD card in the same flight.  Is this possible?

I think the integrated sdcard readers on SPI should be able to handle the extra data even at 2khz logging.

howeson1, 2017-02-01:
I second this, would be very useful to directly compare the differences

fftunes, 2017-02-04:
Disable notch filters for 1 flight instead?

RC-Slater, 2017-02-04:
I suppose that would achieve the same thing, but it would be nice to leave both debug modes operating all the time.

Eganwp, 2017-02-05:
With all the soft mounting, bb'ing and vibration analysis going on these days, I'd propose we leave debug_mode = gyro on by default. Far too many times I see people trying to setup their filters without raw gyro data. 

I could statically assign gyro debug_mode on by default and then still allow the option to enable a second debug_mode. Thoughts?

RC-Slater, 2017-02-05:
Yes I would absolutely like to see that.

mikeller, 2017-02-05:
@Eganwp Sounds like a good suggestion for targets with SDCard or a large onboard flash storage. For targets with smaller storage, or with the constraints of having to log through serial, I wouldn't want to log extra data that cannot be switched off.

RC-Slater, 2017-02-05:
how much extra data would the extra debug stream create?  I can't imagine it would be that much more compared to all the other data streams logged by default.

martinbudden, 2017-02-05:
I am absolutely against having gyro filter logging on by default. The gyro filtering is in a part of the code that extremely time sensitive, especially for fast looptimes or F1 targets.

A solution is to have a debug mode or blackbox setting that allows this recording.

However I think a better solution is to record the raw data and have a setting in the blackbox view that allows you to change the filter parameters so you can see what the data looks like with the filters applied. That is add filter code in the blackbox viewer to simulate the effect of filters. Then you can alter the filter parameters until you have an output that you think looks good and then set your craft to use those parameters and do a test flight.