betaflight/betaflight #2287: bluejay f4 clone fails to arm with 32kz gyro sampling mode enabled

UncleMcNutz, 2017-02-01:
hi issue is present only when the enable gyro 32khz sampling made is on in configuration and led status indicator flashes  unknown what this means as documentation is vague for the clone, not sure if intended at all

borisbstyle, 2017-02-01:
Try increaseing moron_threshold to something like 100

32khz is more noisy and often fails during standard gyro calibration

fftunes, 2017-02-01:
Also be aware, (from what i've seen) the noise could be so bad it may be unflyable.

borisbstyle, 2017-02-01:
Well not necessarily. I have a rig with bluejayf4 and softmounted fc and it really flies fine

ventpolicy, 2017-02-02:
@UncleMcNutz 32khz, aka auto takeoff mode flies like complete shit. don't waste your time. stick to 4 khz for best results.

UncleMcNutz, 2017-02-02:
@borisbstyle not sure if serious with the moron_threshold lol but it's more of when the 32khz switch is active regardless of what gyro or pid i choose even under 8khz it won't arm

martinbudden, 2017-02-02:
@UncleMcNutz , yes he was serious about the `moron_threshold`. This is used during gyro calibration to detect if the user moved the board. However, since the gyro is noisier when using 32kHz, this threshold needs to be set higher to allow the gyro to calibrate. And if the gyro is not calibrated the board won't arm. So have you tried a higher `moron_threshold`?

borisbstyle, 2017-02-02:
@UncleMcNutz The switch activates internal 32khz gyro mode. It doesnt matter if you select 8khz below the gyro is still running 32khz when switch enabled.

EFAW, 2017-02-02:
I have this issue too, although I was just trying 32khz for experimental purposes, 8/8 and D600 is absolutely perfect on 3.1.2.

I tried moron_threshold 100 with no luck, I will try set it to 200 and see if it works.
Does increasing moron threshold effect flying performance adversely in any aspect?

mikeller, 2017-02-03:
@EFAW: No, not at all, as long as the craft is not being moved right after being powered up, which will lead to a faulty calibration if `moron_threshold` is set too high.

If you set `debug_mode = gyro` your blackbox log will report the noise level before calibration in `debug[3]`. You can use this as a guide and set to this value + 20% or so.

EFAW, 2017-02-03:
@mikeller 

Good to know.

I'd love to debug but the BB is a death switch on the BJF4 clone boards, posted the issue below.

https://github.com/betaflight/betaflight/issues/2222

HenryIvarsson, 2017-02-11:
I have two original BlueJays f4 rev 3, both softmounted. Both fly great but only one of them can arm with 32khz.

mikeller, 2017-02-11:
@HenryIvarsson: Read the comments above about `moron_threshold`, and adjust it accordingly to make your craft arm with 32kHz on.

HenryIvarsson, 2017-02-12:
Thank you, will try!

closed 2017-02-21 by blckmn