betaflight/betaflight #2291: Sending RC commands via MSP_SET_RAW_RC and GUI Receiver tab not possible since 3.0.1

christianrauch, 2017-02-02:
I am unable to send RC commands using the `MSP_SET_RAW_RC` message since betaflight 3.0.1 (NAZE).
Up until 2.9.1, after enabling feature `RX_MSP`, I am able to use the GUI control sticks in the Receiver tab to send RC commands and arm the FC.
From 3.0.1 on, with `RX_MSP` enabled, the RC channels aren't visualised anymore in the Receiver tab (`MSP_RC` message seems to be empty), and I am not able to arm via the GUI control sticks.

christianrauch, 2017-02-02:
I found, that one needs to enable this feature by `#define USE_RX_MSP` in the `target.h`.

Is there a reason that this feature is disabled by default (e.g. the messages are accepted but ignored)?

mikeller, 2017-02-02:
Yes. NAZE (or rather: all F1 based boards) is running very low on the flash space available, so all but the most commonly used features have had to be disabled.

christianrauch, 2017-02-02:
I see. Actually, the generic NAZE target linked with `#define USE_RX_MSP` still fits in the flash. I guess this restriction then applies to other F1 targets.

mikeller, 2017-02-02:
Yeah, that, and it's an option that has little value to non-developers, so people (like you) who need it can always enable it and build a custom binary.

closed 2017-02-02 by christianrauch