betaflight/betaflight #2298: Nasty vibration noice BF3.1.1

raymonkroeze, 2017-02-02:
I've tried to upgrade from BF3.0. With BF3.1.1 I have this high pitch oscilation like noice in my quad. With BF3.0 it flies just perfect. With BF3.1.1 it's very noicy! It doesn't seem to appear in the video. Only noice! What can causing this? 
My setup:
spracingF3 like fc
aikon electronics 30a
Tmotor F40II 2400kv
Gemfan masterprops 5*4*3

Running Dshot600

jrwrodgers, 2017-02-02:
Do you have any blackbox logging of the oscillations?

raymonkroeze, 2017-02-02:
No, i don't have blackbox... I'm think it has something to do with the filters... But I'm kinda in the dark here

borisbstyle, 2017-02-02:
If the noise doesnt show in video its your fpv camera. Video should show noise if there is one

raymonkroeze, 2017-02-02:
It's the sound. A jittery nervous sound. and sometimes it oscillates. Not visual. Just the sound. I just switched back to BF3.0 and problem is gone. When I go back to 3.1 it's there again

raymonkroeze, 2017-02-02:
And PID's doesn't seem to have a big influence on this.

borisbstyle, 2017-02-02:
Filter settings are the same between 3.0 and 3.1.

Did you use different filters before?

Also post your dump, but also try 3.1.3 version

raymonkroeze, 2017-02-02:
I never have touched any filters. I do use the Dsetpoint weight and Dsetpoint transition.. I set them a bit lower then default. I also softmounted the motors to give it a try. But that also don't work. 
I'm basically use default setting. As I always do. except for the pids and rates ofcourse. I will try 3.1.3 tomorrow! Motors are new btw! so the still run very smooth. Weird thing is, it's just the sound. In the video/hd i don't see any vibrations...

borisbstyle, 2017-02-02:
Thats indeed weird. Were you also using the same motor protocol?

raymonkroeze, 2017-02-02:
That's different i guess. on 3.1 i use Dshot600 and with BF3.0 i use multishot. I'm running 4/4/32khz

raymonkroeze, 2017-02-02:
I haven't tried to set it to multishot.... I should have tried that too..

borisbstyle, 2017-02-02:
Well probably thats your esc. Do you have caps removed?

raymonkroeze, 2017-02-02:
It's the new all in one esc from aikon electronics! It's not for sale yet

borisbstyle, 2017-02-02:
For dshot600 some escs require to remove the cap.
I removed the cap on my aikon30 and it works like a charm

raymonkroeze, 2017-02-02:
Yeah i know. But this esc should be Dshot ready... It's a flightcontroller size all in one esc...

raymonkroeze, 2017-02-05:
I did some more testing. I tried 3.1.3 and no jitter! so good news. But still having issues.. my lipo is dead more faster than normal. I also set my motortiming to medium instead of medium-high. Don't know if that matters. But after a very fast flip or roll the quad doesn't seem to keep up or something. But not in a way like the Pgain is too low... Any thoughts on this?