betaflight/betaflight #2300: Feature request / discussion: Adjustment of VTX (video transmitter) settings via transmitter aux channels

tigert, 2017-02-02:
I submitted #2293 about automatic adjustment of vtx power via RSSI level and arm state. However sometimes you want to do these things manually (or you might not have telemetry) - thus it would make sense to have this functionality accessible as inflight adjustment / mode too?

Pit mode is kind of obvious as a "mode" you can set via switch, but maybe the power and channel settings could also be done via the inflight adjustment framework?

jflyper, 2017-02-03:
Honestly speaking, I don't like the inflight adjustment feature. I don't know how I can come up with fail safe and fool proof mechanism for the vital function, a sight.

mikeller, 2017-02-03:
Yeah, I'd go for a solution that prevents changes while armed.

mikeller, 2017-02-03:
In saying that, the 'inflight adjustment feature' is probably a misnomer, it should be possible to add a VTX config there that is only applied if the craft is not armed.

andymac4182, 2017-02-03:
Singularity from memory locks the settings once you arm to prevent failsafe issues. It would be good to have the 25mw to full power set up like the Helix and Vortex based on arm state. This helps prevent damage to vTx if you forget to attach antenna when powering on the bench.

tigert, 2017-02-03:
I was mostly thinking allowing adjustment of the VTX power while flying. Changing channels while armed is likely not a good idea.

DeepnineRC, 2017-02-05:
I'm using Pitmode on my TBS Unify. But always going into the OSD menu after power up is annoying.

I like to see an mode switch "VTX_LEAVE_PITMODE" when it is switched to active for one second Pitmode is set off. You can assign an switch on the transmitter to exit pitmode. As security feature: If the mode is longer than e.g. 2 seconds active nothing is changed (if you have forgotton your switch on).

hydra, 2017-02-26:
I added some thoughts on this subject here:
https://github.com/betaflight/betaflight/issues/2392#issuecomment-282563944

cleanflight v1.x already has code to turn a vtx on/off if you need a simple code example.

joshuabardwell, 2017-03-26:
I strongly second @DeepnineRC's comment. Having to go into the menu to exit pit mode is annoying, and it means I don't use pit mode. If I could enable/disable pit mode with an aux switch that would be idea. I could just tie it to arming, or whatever. And I could easily go back into pit mode when I land. Please implement this feature!

digaus, 2017-03-26:
On  arm would be awesome  since when you crash  you disarm and prevent a potential VTX damage if you lost the antenna. Should be an optional feature thought.

tigert, 2017-09-09:
@jflyper I think I actually agree, after some thought. Video feed is something you dont want to be messing with when flying. A glitch at a bad moment is not what you want.

I think the automatic pitmode tied to arming is a good idea though, because like @joshuabardwell said, hard to use pitmode will not be used. But with the following possible pitfall:

At least my Matek VTX has a few seconds delay before it changes settings, probably to avoid intermediate changes when using the buttons to change stuff. This *might* create a situation where you arm, take off and your vtx switches from pitmode to the selected transmission power when you are airborne, causing a momentary video loss. Could perhaps be taken into account with some thought, though, worth thinking because the feature itself would be useful.

hydra, 2018-05-05:
is there a consensus here or can this be closed?

closed 2018-06-11 by stale[bot]