betaflight/betaflight #2301: Changing profiles via stick does not work from 3.1 onwards

emski0, 2017-02-02:
Changing profiles via sticks (like https://github.com/cleanflight/cleanflight/blob/master/docs/Profiles.md) was working before betaflight 3.1

Was the feature removed?
thanks

martinbudden, 2017-02-02:
Yes, I seem to recall this was removed.

mikeller, 2017-02-03:
Do you remember what the reason was for removal? I have never used it, but for people using different profiles this seems to be a useful feature.

MJ666, 2017-02-03:
The 3 profiles are still there? can the be only switched via the configurator or may be OSD? Many people use this to fly 3S or 4S and using individual tunes. Also to have a more relaxed tune and a race tune could be an valid use case. I was using this for thesting diffenet tunes an sswith for and back to compare the easyly. Are we now start stripping down functionality to save space? In my view removing this useful feature does not make real sense.

andre92, 2017-02-03:
Please do not remove this feature. I usually go to the field without laptop but with different batteries/propellers, and I need the ability to switch profiles from transmitter. Or is it possible to switch 3 profiles with one 3pos switch?

MJ666, 2017-02-03:
@emski0  just tested with 3.1.0 on a Naze (only 2 profiles available, space limitations) and a PikoBLX. Everything works as expected for me. Likely you need to verify your stick endpoints and the related RX signals. Looks to be you not reach the conditions for the stick commands if it does not work for you.

emski0, 2017-02-03:
@MJ666 my stick end point are as they should be 1000 1500 2000.
Tried again, i cannot change the profile via stick combination
i am using SP F3 EVO.

andre92, 2017-02-03:
@emski0 I assume your min_check and max_check are default or still between 1000-2000?

borisbstyle, 2017-02-03:
This feature is still in there as far as I know.

However i think it should be better to have it on a adjustment switch.

emski0, 2017-02-03:
i had max_check =1090 instead 1990, typo mistake.
i confirm changing profiles via sticks still works on 3.1.3
However it would be nicer if changing profiles can be done via adjustment switch.
sorry guys and thanks

closed 2017-02-03 by emski0

mikeller, 2017-02-04:
@borisbstyle Totally agree to have it on an adjustment switch. I think the original reason to have it on stick commands was that back then the switch / adjustment switch configuration was part of the profile, so switching from one profile to another could cause not being able to switch back. In addition to that, changing the profile was essentially done by changing the eeprom value for the profile to load, then doing a reboot.

I think we should target moving profiles changing to an adjustment switch (and probably mergeing profiles / rateprofiles into one thing).

Also, I can confirm that profile switching still works with 3.1.

MJ666, 2017-02-04:
Not all TX have enough available switches. I see the value to have this on a switch but the option to change via stick should be still kept. Merging profile and rate profile is a good plan.

Eganwp, 2017-02-04:
100% agree on merging rateprofile with profiles and making it adjust switch assignable. If anyone wants different rate profiles as before, simply use the same pids on that profile and net the same result. I too don't want to boost my channel count above 8 because of 9ms frames, so combing the two would be a huge benefit.