betaflight/betaflight #2305: No PWM output and no motor activity when using PWM ESC protocol

hydra, 2017-02-02:
Steps to repeat:
Flash v3.1.3 firmware, full chip erase.
Use configurator 1.9.1 and change esc protocol to PWM, save reboot.

observe no PWM output on any motor pin on oscilloscope

Additional issue, not sure if it's related.

Go to MOTORS tab.
ARM
Observe motors stay at 1070 when armed, moving board or sticks makes no difference. (unexpected)
DISARM
Observe motors reset to 1000 as expected.

hydra, 2017-02-02:
Target for testing: REVO.

hydra, 2017-02-02:
SPRACINGF3EVO target seems OK.

default motor_pwm_rate on F4 target is 2000
default motor_pwm_rate on F3 target is 400

changing motor_pwm_rate on F4 target to 400 makes PWM signal appear, but PWM signal/motors tab still not working.

wajoli, 2017-02-05:
With the procedure descriped from Hydra:

_Flash v3.1.3 firmware, full chip erase.
Use configurator 1.9.1 and change esc protocol to PWM, save reboot._

on my CC3D I got after arming Full Throttle on motor 3 (see screenshot)

[image]

hydra, 2017-02-05:
CC3D is F1, it seems to be F4 related, thanks for confirmation of an F1 target.

amelluk, 2017-02-05:
This looks to be related to #2306

jflyper, 2017-02-06:
May be related to #2322

Fix has been submitted #2306

borisbstyle, 2017-02-07:
Fixed in 3.1.4

closed 2017-02-07 by borisbstyle