betaflight/betaflight #2308: RCExplorer F3FC resource

fly4props, 2017-02-02:
I have been having issues with the resource  motor mapping on 3.1.0 RCExplorer target , I saw it was fixed in 3.1.2. However the mapping is not right IMO. I loaded 3.1.2 and the default mapping (from the mixer) didn't work. What did work (and how I got 3.1 to work) was this layout; 

resource MOTOR 1 A08
resource MOTOR 2 B00
resource MOTOR 3 A04
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource SERVO 1 A07

I havent tried 3.1.3 as of yet. 

Thanks guys for your hard work

mikeller, 2017-02-03:
From looking at the resources you use (3 motors, 1 servo), I am guessing that you are using the board for a tricopter. The default resource mappings on all boards are for use on a quadcopter. If you want to use the board for a tricopter, you will have to manually remap the resources so you get a resource for the tail servo. Because of MCU resource constraints it is not possible to do this remapping in a generic way - it has to be done in a way that works for the board in question (this is what `pwm_mapping.c` did before 3.1).

Can I ask you to add information about the resource remapping for the RCEXPLORERF3 for use in a tricopter to the board's wiki page? (https://github.com/betaflight/betaflight/wiki/Board---RCEXPLORERF3 ) Thanks.

martinbudden, 2017-02-03:
@mikeller , since rcexplorer.se specialises in tricopters and sells the RCEXPLORERF3 board as part of a tricopter kit, there's an argument that the default resource mappings should be for a tricopter and that it is people who use it for a quadcopter should have to do remapping.

LaserLock, 2017-02-03:
but there are two different versions out

- one is for tricopter (integrated in the frame, https://rcexplorer.se/product/f3fc-tricopter-frame-with-integrated-power-distribution/) 
- one is a simple FC (https://rcexplorer.se/product/f3fc-racing/) useable for a quad

also most of the times he post a dump for cli to make it like it should.
additional there is triflight especially written to work for the tricopters.

martinbudden, 2017-02-03:
> one is a simple FC (https://rcexplorer.se/product/f3fc-racing/) useable for a quad

Yes, but that is sold as part of the baby tri kit. I expect most people using that FC are using it for tricopters.

LaserLock, 2017-02-03:
not exclusively

> Product Description
> Made for Tri and Quad copters.

but thats just my opinion

lichtl, 2017-02-03:
I am not sure if it make sense to split them into like  F3FC-RACE and F3FC-TRI . I would also thought that the F3FC is mainly used on the tri but agree it can be used for a quad as well while the frame I think is an exclusive Tri FC.

fly4props, 2017-02-03:
@mikeller Ill write something up for the wiki today. Thanks for your imput. I just assumed when you set the mixer for a tricopter the firmware  would automatically remap the motor outputs.

closed 2017-02-03 by fly4props

mikeller, 2017-02-04:
@martinbudden I wasn't aware that they are selling the board as part of a tricopter kit, and neither that there is a version of it that comes as an FC integrated baseplate.
In this case there is indeed a point to be made for having this board default to having resources allocated for tricopter, and the tricopter mixer selected by default. But I am not sure if this is the right thing to do, since I think that tricopter users are probably a bit more accustomed to having to do resource remapping for their builds to work, whereas somebody wanting to use the board for a quad will likely attempt to just change the mixer to quad, and then not get all motors to work because of the resource mapping.

AndersHoglund, 2017-02-04:
Seems as this would require YAT (Yet Another Target), RCEXPLORERF_TRI and perhaps RCEXPLORER_QUAD etc etc in the same way RF releases binaries for different configs ( _STD, _SPEKTRUM_TELEMTRY etc) But this is not manageable in the long run. 

Or the generic binary together with pre-made config files for TRI etc etc QUAD in the way it has been  envisioned earlier. Almost equally hard to manage, without tooling support. Random config hints on RCG, FB, slack or in a Wiki will not do, there must be a more systematic and structured way. 

> I just assumed when you set the mixer for a tricopter the firmware would automatically remap the motor outputs.

Me too, but it seems the resource management is very static, not dynamic in the way you would expect as a user. Same assumption and expectations really as for SERVO_TILT, CHANNEL_FWD, PWM, LED_STRIP etc etc

mikeller, 2017-02-04:
No, NMNUT! (No more unnecessary targets)

LaserLock, 2017-02-05:
change the resource mapping on the quad / tri dropdown?

mikeller, 2017-02-05:
If it only was that easy. Unfortunately there is no universal way to change the mapping, as what pins will work for what depends on constraints inherent to the MCU. So in order to make this automatic remapping work, there would have to be a database of all possible supported mappings. (and factors like OneShot or DShot being used, or PPM or serial Rx will influence what is possible and what not) for every target.

martinbudden, 2017-02-05:
I'm not suggesting we have two targets for the RCEXPLORER FC, merely that the default configuration is for tricopter. If (say) 90% of the people who use this board put it in a tricopter then that makes sense. If a lot of people use it in quadcopters, then it's not such a good idea.

My suspicion is that very few people put this FC in a quadcopter. But rather than debate this among ourselves, I suggest we ask David Windestål of rcexplorer.se , the designer of the board (and the tricopters that he sells as well) as to what his preference is.

martinbudden, 2017-02-05:
OK I've sent David Windestål a message. Let's see what he says.

RCExplorer-David, 2017-02-09:
Sorry about the delay,
I would suggest doing the Quad config the standard. The reasoning being that most people that get the FC to use on a tricopter will use Triflight and those who use it on a Quad will flash Betaflight most likely.
Thoughts?

mikeller, 2017-02-09:
I think that's a fair point, @RCExplorer-David.

martinbudden, 2017-02-09:
@RCExplorer-David , I wasn't sure what people use, that's why I asked you :-)

So setting betaflight defaults for quad it is. It's only the defaults, so it doesn't prevent people changing the setup.