betaflight/betaflight #2309: AIR32 - DSHOT not work (working solution)

Szpero, 2017-02-03:
AIR32 not using DMA for motors when DSHOT is active.
Culprint is target.c

Working code:
#include <stdint.h>

#include <platform.h>
#include "drivers/io.h"

#include "drivers/timer.h"
#include "drivers/timer_def.h"
#include "drivers/dma.h"

const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
	DEF_TIM(TIM3,  CH2, PA4, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM3,  CH1, PA6, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM3,  CH3, PB0, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM3,  CH4, PB1, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM2,  CH2, PA1, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM2,  CH3, PA2, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM15, CH2, PA3, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM1,  CH1, PA8, TIM_USE_MOTOR, 1),
	DEF_TIM(TIM17, CH1, PA7, TIM_USE_PPM, 0),
	DEF_TIM(TIM16, CH1, PB8, TIM_USE_LED, 0),
};

mikeller, 2017-02-03:
Thanks for working this out. Can you please submit a pull request for this change?

Szpero, 2017-02-03:
Commited.
I leaved stock timer and channel assigment. Still something need to be remaped to have all motors working with DSHOT - for example PPM PA7 to motor 1.

mikeller, 2017-02-03:
The target definition should match the pin assignments as they are printed on the board / in the manual - and they normally work fine this way if DShot isn't used.

But please add your instructions on how to make DShot work on the AIR32 to the DShot wiki page:
https://github.com/betaflight/betaflight/wiki/Board---AIR32

Also, thank you very much for the PR.

Szpero, 2017-02-08:
Should I create in AIR32 page (your link) or here should be enough [https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol](https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol) ?
As I see AIR32 using same principles like PIKOBLX. So in my opinion best approach to remap is:
```
resource motor 5 free
resource motor 6 free
resource motor 1 A01
resource motor 2 A02
save
```
Motor 1 solder to motor 5 pad on FC.
Motor 2 and 3 leave as is.
Motor 6 solder to motor 6 pad on FC.

LED output should working - its sharing DMA with motor 4 output.
PPM pad sharing same DMA channel like MOT6 so taking PPM pad for one of motors doesn't make sense. lets leave motors on dedicated pads - MOT1-6.

Performance (tested on desk):
4k/4k acc on, led off, frsky telemetry on, msp for OSD, DS600 - ~11% CPU
8k/8k acc on, led off, frsky telemetry on, msp for OSD, DS600 - ~56% CPU - little to high in my opinion.
8k/8k acc off, led off, frsky telemetry on, msp for OSD, DS600 - ~32% CPU - should be fine. Didnt tested.

waltr1, 2017-02-08:
Szpero
What I did for the MotoLab FCs is have the instructions on how to re-mpa and rewire to make DSHOT work in the DSHOT wiki page.

Then on the MOTOLAB page list this under FAQ & Known Issues with a link to the DSHOT page.

Every FC Target/board should have a wiki page with all basic info. Please do start a page for the AIR32. There is a template that should help getting the new page setup.
Thanks

Szpero, 2017-02-27:
Thanks for help waltr1. I made some description about AIR32 and I hope it will be sufficient.
Here is movie from in open air test - DSHOT working for now fine (master branch 3.2.0):
https://www.youtube.com/watch?v=4o6JOLC30Js Not well tuned quad and I still need to improve piloting skills :) .
Today I replaced software with more realiable version: 3.1.6 + DSHOT target.c . On desk its working. Now I need to find some time for terrain tests.

Szpero, 2017-05-08:
Closing case.
So far tested a lot AIR32 with adjusted target file and compiled by self 3.1.6 version.
DSHOT working fine.
AIR32 8k/8k DSHOT600 (remap), FVT 30A-S (decap)

closed 2017-05-08 by Szpero

mikeller, 2017-05-09:
@Szpero: Thanks for adding the documentation to the wiki. I've added an entry to the DShot page, with a link to the FC page.