betaflight/betaflight #2313: OSD artificial Horizon is going in the wrong direction on roll

dschiedsch, 2017-02-03:
While rolling the aircraft the artificial horizon is moving the same way as the aircraft  not the other direction as it should be. (artificial horizon should move with real horizon not in the other direction)
I corrected this by changing
int y = (rollAngle * x) / 64;    to
int y = (-rollAngle * x) / 64;
in osd.c line 316 in the current master branch.
Don't know if this is the correct way to fix it that's why I didn't upload my code.
Tested it working.
My used target is OMNIBUS 
on the Eachine AIOF3_BRUSHED Flight Control Board Built-in OSD

Mikowmer, 2017-02-04:
Could it be that the gyro/accelerometer is the wrong way around? In which case, wouldn't it be a better idea to add something to invert the roll direction if needed?

To test the first thing, is it rolling the correct way in the configurator when you connect the aircraft to the computer?

AndersHoglund, 2017-02-04:
Depends if you want Western or Russian standard.
http://www.skybrary.aero/images/thumb/Artificial_Horizon_Formats.jpg/400px-Artificial_Horizon_Formats.jpg

fftunes, 2017-02-05:
@AndersHoglund Are you serious?

mikeller, 2017-02-05:
Looks like something the russians would do. :-P

AndersHoglund, 2017-02-05:
Serious indeed. There have even been incidents/crashes with Russian pilots in Western aircrafts.
http://www.aviationtoday.com/2005/07/01/safety-mode-confusion-timidity-factors/

tigert, 2017-02-06:
https://www.youtube.com/watch?v=fvFZGkerIj8 - Mig 29 HUD

This feels weird when one is used to the western artificial horizon / HUD, but I can see it works, its just a different way of displaying the same information. If you are used to one, the other one is more or less equally confusing to you, I guess. Regardless of which one it is.

However, the russian style needs a larger "airplane symbol", especially given the character-based HUD technology. A vector OSD would be cool, in other ways too.

macht-nix, 2017-02-07:
same Poblem here on OMNIBUSF4SD
plus it freezes at an angle over 45° (pitch and roll)
plus no sidebars if the Horizon is off
There is still a lot to do to get the usability of eg "KV-OSD" what i used on my oder Builds

[https://youtu.be/ehp-pBIyS_Y](url)

zac59, 2017-04-10:
same Poblem here on OMNIBUS F3

xLizrd, 2017-04-21:
Same problem on the Omnibus F4.

jflyper, 2017-10-23:
We need a switch on OSD so that users can select

- Russian
- Everywhere else

closed 2017-10-23 by jflyper

mikeller, 2017-10-23:
That is provided that there are > 0 users who actually want russian (also, we should convert the OSD text to cyrillic in this case). :grin: