betaflight/betaflight #2314: CC3D Revo F4 (target: AIRBOTF4) current meter not working with 3.1.x

nhadrian, 2017-02-03:
Hi all, I have just moved to an [F4 controller](http://www.banggood.com/Raceflight-Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-with-VbatBuzzer-p-1087561.html?p=5A28025516662014049E). I'm using a  [BeeRotor PDB](http://rctimer.com/product-1619.html).
Previously I used this PDB along with a BeeRotor F3 controller, both current and voltage monitoring worked well. With this new F4 board I have to use the built in voltage measurement with the on-board BEC because it doesn't have input for external voltage monitoring. This part is working.
But there is an input for current measurement. I configured everything but it looks like that current measurement is not working, values near 0,0A are shown despite any amperage.

Could anybody test and confirm a working AIRBOTF4 current measurement?

Regards
Adrian

mikeller, 2017-02-04:
You need to calibrate your setup first: https://github.com/cleanflight/cleanflight/blob/master/docs/Battery.md If you hit the limits of the calibration range without being able to calibrate this means that the output of the BEC's current sensing pin is not within the range of what your board expects on this pin.

nhadrian, 2017-02-04:
Hi, of course, I know those.
I use the same amperage_meter_scale amperage_meter_offset values that I used on the other F3 board, those values were carefully calibrated. So it should work on this F4 panel too I think, maybe some minor fine tuning would be neccessary. But current meter doesn't working at all, doesn't matter what amperage are going through the sensor, the readed value is around 0,0 A.

mikeller, 2017-02-04:
That it worked with the hardware on one board doesn't mean it's right for the hardware on a different board. There is no standardised way for how these current sensing circuits are built, and different boards use different circuits with different dividers. So the right way to go on about this is to ignore the values from your old board, and do a new calibration with your new setup, and see if that works (it may not).

nhadrian, 2017-02-04:
Unfortunatelly doesn't work.
Modifying the offset also moves the idle value + or - but the number doesn't change on amperage increase. I also tried several variations of scale from 100 to 400, nothing happens.

blckmn, 2017-02-21:
I'd find out what pin the current sensor is attached to. Then run `resource` in the command line to verify it is setup correctly.