betaflight/betaflight #2315: BF 3.1 - SPRACINGF3 - PWM RX input - Channels screwed up

mgaller, 2017-02-04:
On betaflight 3.1 i only get 4 Channels moving in the reciever Tab instead of 6.
Only Aux 1 - Aux 4 is moving (when i move throttle or so), no movement on Roll, Pitch, Yaw or Throttle.
On Betaflight 3.0.1 it worked fine. 
Reflashed 3.0.1 and its working again, so my wiring is not the problem.
Flashed back to 3.1 and only enabled PWM RX input and the same problem occurs.

Diff of 3.0.1
```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# BetaFlight/SPRACINGF3 3.0.1 Oct 18 2016 / 10:35:52 (48b7b4f)

# name

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_PARALLEL_PWM
feature AIRMODE

# beeper

# map
map RETA1234

# serial
serial 2 1 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1825 2100
aux 1 2 0 1300 2100
aux 2 12 0 1875 2100

# adjrange

# rxrange

# rxfail

# master
set min_throttle = 1050
set max_throttle = 1980
set use_unsynced_pwm = ON
set motor_pwm_rate = 2000
set align_board_yaw = 90
set deadband = 2
set yaw_deadband = 2
set failsafe_delay = 20

# profile
profile 0

set p_pitch = 70
set d_pitch = 18
set p_roll = 40
set d_roll = 16
rateprofile 0

# rateprofile
rateprofile 0

# 
```
Diff for 3.1
```

CLI

# diff
# Betaflight / SPRACINGF3 3.1.3 Feb  2 2017 / 13:58:33 (73b498e)

feature -RX_PPM
feature -VBAT
feature -TELEMETRY
feature -CURRENT_METER
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_PARALLEL_PWM
feature AIRMODE
map RETA1234
serial 2 1 115200 57600 0 115200
aux 0 0 1 1825 2100
aux 1 2 0 1300 2100
aux 2 12 0 1875 2100
set min_throttle = 1050
set max_throttle = 1980
set align_board_yaw = 90
set gyro_sync_denom = 2
set deadband = 2
set yaw_deadband = 2
profile 0

set d_pitch = 20
set d_roll = 18
rateprofile 0
rateprofile 0

# 
```

I got another Quad with x racer F303 v3.1, iam gonna try it there too.

mgaller, 2017-02-04:
Tried it on x racer F303 v3.1, problem does not occur. (same reciever)

blckmn, 2017-02-21:
They both use the same target SPRACINGF3 correct?

Can you post the resource command for both boards?

mgaller, 2017-02-21:
No, SPRacing F3 is using target SPRACINGF3 and x racer F303 v3.1 is X_RACERSPI
I didn't do any resource changes.

SPRacing F3 
```

# resource
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08

# resource list
IO
A00: PWM 1
A01: PWM 2
A02: FREE 
A03: FREE 
A04: FREE 
A05: FREE 
A06: MOTOR 1
A07: MOTOR 2
A08: FREE 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 3
A12: MOTOR 4
A13: FREE 
A14: FREE 
A15: FREE 
B00: PWM 7
B01: PWM 8
B02: ADC_RSSI 
B03: LED 1
B04: PWM 5
B05: PWM 6
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE 
B09: FREE 
B10: PWM 4
B11: PWM 3
B12: FLASH_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: FREE 
C15: BEEPER 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

```

x racer f303 v3.1

```
CLI

# resource
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 A08
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B00
resource PWM 6 B01
resource LED_STRIP 1 A08

# resource list
IO
A00: PWM 1
A01: PWM 2
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE 
A05: FREE 
A06: MOTOR 1
A07: MOTOR 2
A08: FREE 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 3
A12: MOTOR 4
A13: FREE 
A14: FREE 
A15: MPU_CS 
B00: PWM 5
B01: PWM 6
B02: FREE 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE 
B09: FREE 
B10: PWM 4
B11: PWM 3
B12: FLASH_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: LED 1
C15: BEEPER 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: FREE
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

# 
```

djwally, 2017-07-15:
i have the same problem, but for me not working on any firmware in betaflight and cleanflight. 
You solved it?

digo0170, 2017-08-04:
I have the same problem too.
When I put a ppm rx works perfectly.
It turns out I have a few pwm rx left, and would like to put it to use.
Has anyone managed to solve the problem of using RX PWM in BetaFlight 3.1.x (top)?

jflyper, 2017-08-05:
I think PWM RX input is screwed for all targets. I had a slack support session with NAZE recently who has the exact same issue; first 4 RC/PWM inputs not working at all.

Noctaro, 2017-08-05:
The problem is, 3.01 was the last version with pwm timer clash protection. As far as i know. Thats why PWM rc input is screwed up.