betaflight/betaflight #2319: BF 3.1.3 SPRACINGF3 (Diatone D-Link V1) no longer working with BLHeliSuite

FliteBrothers, 2017-02-04:
I flashed my ESCS (Diatone X302 aka DYS XMS20 clones) with BlHeli 16.6 on BlHeliSuite 16.6.14.9.0.0 using pass-through while still on BF 2.9.1 and worked flawlessly.

I then flashed the FC with 3.1.3 and since then, I am no longer able to connect to the ESC's using BlHeliSuite.

AndersHoglund, 2017-02-04:
I can access my BlHeli16.65 ESCs thru SPRF3 BF3.1.3, provided I turn of DSHOT first. With MULTISHOT selected, there is no problem. With Dshot300 selected I get this on connect:

[image]

This is also related to `min_command`, I happened to have `min_command = 980` in the tests above. (Remainder from an old MS config.) Changing `min_command = 1000`, and still keeping DSHOT300 selected, also seems to "fix" the problem.  

I would suggest this is a problem with BLHeliSuite, not a Betaflight Firmware problem. Use BLHeli Configurator instead, it does not have this problem.

mikeller, 2017-02-04:
I thought this was fixed at some stage. @DieHertz would know more.

Eganwp, 2017-02-05:
I thought this was always the case because I know all the DShot guides written always have said to choose MS or OneShot before opening Blheli suite or configurator or passthrough doesn't work. I thought this was still the case as I've never seen a PR on this.

AndersHoglund, 2017-02-05:
I am not even sure if this is the OP original problem, there are no symptoms presented other than:

> ...no longer able to connect...

Not much to work with, other than guessing and assuming it is the same as what I see.
Please @FliteBrothers , can you confirm it is the same problem. Or add some more details to the case.

FliteBrothers, 2017-02-05:
No this is not that issue. I had hoped but I was wrong. It may be an issue specifically related to this specific FC which is the Diatone D-Link F3 V1 that flashes Betaflight using the SPRACINGF3 target. This FC was paired with these ESCs on their Crusader GT2 PNF models. Many people who have been upgrading from the stock 2.9.1 BF target to any of the 3.x versions are experiencing connectivity issues that are frequently leading to bricking the ESCS. The problem is occurring after flashing to 16.6 BlHeli. Here is what mine does now even after switching to MultiShot again. I'm now unsure if this is a BetaFlight or ESC issue.
[image]

FliteBrothers, 2017-02-05:
Update: Reading through some of the reports, it appears that it occurs primarily when trying to use 3.1 variants of BF regardless of the current ESC firmware.

FliteBrothers, 2017-02-05:
Confirmed: Flashing Back to 2.9.1 allowed me to connect to my ESCs with BlHeliSuite as normal.

FliteBrothers, 2017-02-06:
Still gathering information from others experiencing the problem, but it appears to occur with any variant of BF 3.1.x when attempting to flash BlHeliS ESC's to 16.6 only. Some reports like mine that they simply could no longer connect or had issues connecting to ESCs after flashing BF 3.1.x
Both scenarios have led to complete bricking of the ESC during the flashing process.

AndersHoglund, 2017-02-06:
Sorry, can not reproduce this problem with my FCs and ESCs. A few hints:
Signal fidelity. You do have both signal and gnd connected to all ESCs?
ESC input filtering, are there any resistor/capacitor on the ESC inputs or FC outputs?

automatt1c, 2017-02-06:
I am having the same problem as well. flashed to 3.1.3. Only ESC 2,3,4  were showing up in BLheli even though the motors worked fine. ESC would occasionally disapear when i tried to read it. After trying to flash the latest version one of the ESCs appeared to be bricked. then i got an unknown ESC on ESC 1 then tried to flash that and that now is bricked as well. All of the solder connection are very clean and no resistors or capacitors that i can see. now even on 2.9.1 i still cannot connect to the two ESCs.

FliteBrothers, 2017-02-06:
@automatt1c Can you verify that you are using a Diatone D-Link FC or something else?

FliteBrothers, 2017-02-06:
@AndersHoglund It does not appear that there is an resistor/cap filtering on the ESC inputs or FC outputs. On this particular FC which is an AIO there is no spot specifically for a ESC signal ground to be connected so they are typically run without it. Would seem unusual that it works perfectly this way on 2.91 and 3.0.1, but not on newer versions of BF though right? I have requested a test using signal ground attached though to see if it works that way.

automatt1c, 2017-02-07:
Yes, I am using the diatone dlink FC. Which uses the spf3 firmware.

AndersHoglund, 2017-02-07:
Oooh, an AIO. Probably with same gnd plane for both motor power and small signalling. I am not sure it is a good idea to mix those two.... 
No signal gnd solders pad would probably matter less in this case. But it is still a bad design, in principal. A signal always needs reference or return path, preferably not shared with high and pulsing power currents. And it is always a good idea to twist ESC signal and gnd. Try to find some signal-gnd solder points close to signal pads, as far away from the power gnd pads as possible. But as you say, BF versions should not matter then.... Worth a try to connect and twist signal gnds anyhow.

gregm412, 2017-02-26:
Has there been any progress on coming back from this? I'm also left with 2 non working esc's. Can anyone point me in the right direction. I just ordered a arduino nano. Thought maybe the esc boot loader is corrupt. I have no idea. Lol please help.

Eganwp, 2017-02-27:
Hi Greg,
I had to fix 3 ESCs because of corruption/failed flash using blheli_config Chrome app recently. Bought an Arduino Uno and fixed them all first try. The best thing to do if you're confused is download blheli_suite and open up the included documentation and study flashing via 4 wire method. It tells you all you'll need for flashing and using your Arduino Nano including pics. :)

- Travis

automatt1c, 2017-02-27:
I bought a ESC flasher but i still couldn't get it to flash or read the
ESCs. I ended up sending the drone back and they are replacing it (when
stock is available). I heard that Diatone said not to upgrade to 3.1. Its
good to hear that someone was able to fix the ESC with a Adruino.

spuder, 2017-02-28:
From my conversations with Diatone support, they have seen a lot of problems with 3.1 and the [diatone d-link F3 v2 flight controller](http://www.diatoneusa.com/store/p374/Diatone_D-Link_F3_V2_PDB). They have gone as far as to hold all shipping quads back at betaflight 3.0 and printing warnings to not upgrade to 3.1. Upgrading will void the warranty and has a high likelyhood of bricking ESCs. 

Here is the package for my new crusader gt2

[image]

Without access to test hardware and an oscilloscope, looking for the code regression might be the most fruitful option. 

Filtering github merge requests to those between September 11th 2016 (3.0.0) and January 25th 2017 (3.1.0) returns:

34 merge requests related to esc's

https://github.com/betaflight/betaflight/issues?utf8=%E2%9C%93&q=%20is%3Amerged%20esc%20merged%3A2016-09-11..2017-01-25%20

9 merge requests related to esc passthrough

https://github.com/betaflight/betaflight/issues?utf8=%E2%9C%93&q=%20is%3Amerged%20passthrough%20merged%3A2016-09-11..2017-01-25%20

## Additional anecdotes for affected quads

- Flight controller - http://www.diatoneusa.com/store/p374/Diatone_D-Link_F3_V2_PDB
- Esc - http://www.diatoneusa.com/store/p239/DIATONE_302X_BLHELI_S_30A_ESC
- Signal ground not connected
- ESC bricking seems random. Some work, some brick on upgrading to 3.1.0

spuder, 2017-02-28:
This issue smells very related https://github.com/betaflight/betaflight/pull/1815

borisbstyle, 2017-02-28:
Why does this work on original Spracingf3 with regular blheli ESC's?

mikeller, 2017-02-28:
@borisbstyle I've been wondering the same. I wonder if their hardware does require some weird config, and they are shipping their hardware with this config. Does anybody have a Diatone D-Link V1 that has not had a firmware update and that still has a mostly intact original config, and if so, could they please post the output of `dump` and `resource list` here?

gregm412, 2017-03-01:
This is just a thought maybe the dlink is not an exact match to the spf3 but running it's code anyhow.

thrasher0815, 2017-03-10:
From what I read on this thread, this seems to corrupt the ESCs only if you upgraded to BF 3.1.x AND tried to flash BLHeli_S. Do BLHeli_S ESCs work in general with the D-Link V1/2 on BF 3.1.x? Do they run Dshot 600 fine? Can they still be configured? Or is upgrading the stock FW corrupting passthrough in any combination? Just upgraded the FW and planned on getting some of the new LittleBee Spring ESCs :/

FliteBrothers, 2017-03-10:
They will operate fine, the issues only happen when trying to connect to
the ESCs for flashing or changing settings. Sometimes they will not even
show up in BLHeliSuite and if they do, even changing settings seems to be
enough to damage the ESC firmware though most reports have been that they
bricked while flashing, not just changing settings.

kinucris, 2017-03-14:
My third flight ...
Latest version of Beta
ESC updated to latest version of BLHeli

After the flight, I landed normal ... I placed the drone on the pavement and after a few seconds the same caught fire in one of the ESC!

[image]

iPilotx, 2017-04-26:
Hi,

Just to be clear, If I remove the stock ESC's and install some other brand, I may then upgrade my D-link V2 to 3.1.x or am I still bound to 3.0.1 and lower if i'd like to maintain any sort of stability ? I've had my ESC's brick up and bought 4 new ones of the same brand. But if going to a different brand allows me to upgrade to 3.1.x then I'd like to move on from my X302s. From what I understand however it's got to do with the FC not allowing pass through if updated any further then 3.0.1? 

This is me attempting to get all the info supplied by you guys right as I'd like to make purchases which wont be a waste of money in the long run.

Thanks

gregm412, 2017-04-26:
I installed racestar v2 30amp esc's and updated them from 3.1.7 betaflight and it worked fine for me.

kinucris, 2017-04-26:
Yeah ... maybe the problem is with the ESC ...
Too bad I only figured it out after having a stock of those damn ESC and
after having burned my Original FC and now need a new one.

I think of putting the Betaflight F3

spuder, 2017-05-02:
I upgraded my Diatone GT2 (v2) board to version 3.1.7. 
Now I find that if I open bl_heli configurator, is randomly detects between 1 and 2 of my 4 esc's. 

dump 3.1.7

https://gist.github.com/spuder/2fc3202089349cbdc86d46efd23ac2d9#file-spencers-3-1-md

dump 3.0.1

https://gist.github.com/spuder/2fc3202089349cbdc86d46efd23ac2d9#file-spencers-md

[image]

kinucris, 2017-05-07:
> d9#file-spencers-3-1-md
> d9#file-spencers-md

OMG!

Why the solution?

Chance all four esc? :'(
I have 7 of this (sh#t)

Eganwp, 2017-05-08:
This is highly documented issue on here, it's not just your brand of ESCs having the issue.

iPilotx, 2017-05-08:
So I've been having  look, is this a Betaflight issue ? 

seems like the pass through isn't working very well with newer version of Betaflight from what I can see.

ptuxbury, 2017-06-26:
I'll add my 2 cents here. I had the same issue here with Diatone v2 board. I upgraded the FC to 3.1.6, and then tried to flash the ESCs to BLHeli 3.1.6 via passthrough. One of my ESCs was corrupted and I got the either message ended up having to reflash it through

ptuxbury, 2017-06-26:
Sorry... typing on phone and accidentally hit "comment" button!
One of my ESCs bootloader was corrupted and I got the BLHeli error message. I ended up flashing through C2 bootloader pads. I then downgraded to Betaflight 3.0.1, and was able to flash to BLHeli 16.6 via passthrough with no issues. After I upgraded BLHeli, I then upgraded back to Betaflight 3.1.6 and was able to fly with no problem (DShot).

Of curious note, I recently had the exact thing happen to me on an Omnibus F4 Pro with BF 3.1.7. I'm now making it a process to downgrade to BF 3.0.1 to update ESCs, and then flashing back to the current version of BF after ESC update is completed.