betaflight/betaflight #2320: Death rolls on 3.1.3, possibly earlier

Cheredanine, 2017-02-05:
Had this myself, multiple others reporting it, may be in earlier point releases also.

In my case during single fast roll the quad made no attempt to stop rolling and just spun at full speed until it hit the ground
Not normal desync a la min throttle too low on PWM protocols where the quad attempts to stop spinning then one motor desyncs and quad spins towards that corner, instead quad literally made no attempt to stop and carried on spinning full speed until planet earth stopped it, I had it twice, setup:
Bluejay f4, bf 3.1.3 8khz, 4khz, dshot600 spedix 25a flashed with 16.6, digital idle 4% (motors spin at about 1009)

Other posts can be found at or about page 2964 on rc groups thread, this one has a bb log (although can't gaurentee it is the same, it does sound like it) using other kit including f3 fc:
https://www.rcgroups.com/forums/showpost.php?p=36809146&postcount=44457

Cheredanine, 2017-02-05:
Ps unfortunately didn't have sd card in so bb log for my events but was able to replicate goggles off/horizon mode. Definatley NOT single motor desync, the quad continues to roll at the same speed when command was removed, no other change in aspect, driving motor pair continued to push at the same speed (both visually and audibly)

Aileron8, 2017-02-05:
Could the digital idle be too low? I know a bunch of folks had to raise their digital idle past default recently with BF3.1.3 to ensure their min_throttle was >=1050 in the motors tab. For example, the default 4.5 would certainly cause a desync on my LB20_S ESC's, mimicking a death roll if it only happened to a single motor. I had to increase the digital idle to 6 to enable my min_throttle to reach 1050 in the motors tab.

Cheredanine, 2017-02-05:
@Aileron8 
No mate, two reasons, firstly when min throttle/idle is too low the quad makes an attempt to stop, hen one of the motors desync and the quad tips to that corner, typically it recovers after about 69 degrees, this is not the behaviour I described, it made no attempt to stop and it continued with its roll along the same axis it was already rolling on without dropping the power to the two pushing motors. That was in part why I reproduced in horizon mode, the auto levelling would have made the pushing motors oscillate from low to high power as it rolled, it would have been very apparent. Instead they continues at full thrust
Secondly as per first post, motors spin at 1009 and idle set to 1040

adamtfc87, 2017-02-05:
My drop out of the sky happened with 3.1.1. Only happened once but it was with the x-racer v3.1. Flew 15 packs since and no issues but had no indicators of low RSSI or lost signal. Taranis connection was still there when it did happen.

borisbstyle, 2017-02-05:
The screenshot in related topic shows motor1 failing.

Cheredanine, 2017-02-05:
Then it isn't the same, just repeated with BB, will load logs when I get back from field

adamtfc87, 2017-02-05:
Could this have to do with the CPU% spiking like QuadMcFly suggested? Interesting as this is, I've had 0 problems with this with my BFF3 running 3.1.3 at 8k/8k. Does this issue only happen when people are doing blackbox logs?

Cheredanine, 2017-02-05:
log file, final roll that ends the flight
https://www.dropbox.com/s/9doyiq3tkywoojp/LOG00001.BFL?dl=0

Cheredanine, 2017-02-05:
interestingly contrary to what the gyro seems to think, the qud spun left and never stopped, the gyro reading saying it is spinning right is erroneous
the two right hand motors that appear, on the log, to be providing no thrust continued to provide thrust as the log indicates they should be, even though the gyro says it is moving in the opposite direction

Cheredanine, 2017-02-05:
Ok wasn't able to replicate on other quads with different or same fc, had to assume this is physical, not sw based, closing

closed 2017-02-05 by Cheredanine

Oriolkun, 2017-03-24:
Hello I also have the same isssue fo death-rolls, In my case it happnes when I make a fast flip, at any direction. 
@Cheredanine I have seen your pic. I have the same issue as you. Same behaviour. The picture shows something interesting. Just at the moment you rise pitch sitick up P of PID reaches a top level and looks like it saturates. Then when you end the flip and returns to non-saturated pitch P of PID the aircraft starts the deathroll. Juslt what happens to me. Try to low P of pitch value -10 of Betaflight P-pitch default and try to fly again.
I think deathrolls happens because P of pitch get saturated when a fast pitch movement and get over PID limits.

Cheredanine, 2017-03-24:
@Oriolkun 
I replaced my fc with another of the same type, set up exactly the same and it is fine mate, hardware error

Oriolkun, 2017-03-24:
@Cheredanine same FC?, I have the same issue of this  on Airbot F4, Omnibus F4 (no SD), and SPracingF3 (multishot). The issue is the same you show on the pic. When did an aggressive flip start rolling as the black box shows. 
A part from that, in your pic, when the flip starts the motor 3 do not rise, only motor 1. And for doning the flip the motor 3 should rise as motor 1 as well. Can this be the cause of the issue? 

By other hand I'm concerned if BF parameters like  TPA or vBat PID compensation can be causing this issue. Do you have them disabled?

Cheredanine, 2017-03-24:
@Oriolkun 
A new fc of the same type
On a front flip (nose down) motors 1 and 3 should spool up, if only motor 1 spooled the quad would spin diagonally

You do realise this issue is closed, you will not get further support without opening a new one or posting on rcgroups ?

maciek01, 2017-05-21:
i have the same issue: qav250, BFF3, BF 3.1.6, multishot on 20Alittlebees

vinz486, 2017-05-29:
FurousFPV Racewhoop FC, BF 3.2 (custom build form Furious): same issue. When the stick is at maximum, rolls never stops like you do not turn back the stick.