betaflight/betaflight #2325: Timer question

jflyper, 2017-02-05:
Need some help reading a timer code.

This is a piece of code in serial_softserial.c that sets up a bit clock interrupt (TX side timer).

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/serial_softserial.c#L141-L142
```
    timerChCCHandlerInit(&softSerialPorts[reference].timerCb, onSerialTimer);
    timerChConfigCallbacks(timerHardwarePtr, &softSerialPorts[reference].timerCb, NULL);
```

I'm not sure why this is working (in some way or another), because `timerChConfigCallbacks` takes two callback, one for edge interrupt and another for overflow, and the callback set here is for edge. It should be the overflow callback here.

Thoughts?

jflyper, 2017-02-06:
@ledvinap I think this is the reason it did not interrupt on TIM7 (and TIM6).

blckmn, 2017-02-06:
@jflyper have you tried moving it? it certainly seems like an error.

jflyper, 2017-02-06:
@blckmn The code is from the current working code (at least for F3), and wondering why it's working. My PR #2317 no longer uses this code for bit clock (uses overflow).

ledvinap, 2017-02-06:
@jflyper : IMO compare channel will still trigger edge interrupt, but I can't find code where CCxE bit is set.

Overflow callback should be used.

closed 2017-10-23 by jflyper