betaflight/betaflight #2327: Revolt F4 - SBUS and Smartport on same serial port

FirebirdDP, 2017-02-05:
Is there any way to select both **Serial RX** and **Smartport** telemetry on the same serial?  Currently the configurator is only allowing one.

I have wired my Raceflight Revolt F4 as per the instructions:
- SBUS => SBUS (FC/RX)
- Smartport (X4R invert hack) => Smartport (FC/TX)
- Both are on UART 1 

At this moment I can only make one of these work by selecting Serial RX or Smartport on the ports page in the Betaflight configurator.  (Flight controller info, identifier: BTFL, version: 3.1.3 / Board: RVLT, version: 0)

I have not tested this with the Raceflight configurator but I guess they have done so in the firmware.

Thanks

DerickP

MennoBr, 2017-02-05:
I don't think this can be done. Smartport is bidirectional and uses both pins on an uart to communicate although you are only connecting one wire.

geofrancis, 2017-02-06:
also sbus and smartport run at different speeds.

SoulPictures, 2017-02-06:
if i remember it correctly they have kind a weird layout there. The TX pin beneath the SBUS pin is a different UART than the RX SBUS pin

closed 2017-02-06 by borisbstyle