betaflight/betaflight #2328: 3.1 won't spin up motors from PPM receiver

spatzengr, 2017-02-05:
I have a Spektrum receive and am communicating with a SPracerF3 firmware flight controller via. PPM.  I could spin-up the motors through the Motors tab, but not via. the transmitter.  Also, I'm not sure if restricting the PID loop to less than what can be transmitted via. the ESC communication protocol (as 3.1 adds a new restriction) is NOT without issues.  See https://www.screencast.com/t/VkmqtUx6I for my experience in 3.0 when i turned the PID loop down to 0.5kHz after I switched back.

jflyper, 2017-02-06:
May be related to #2322 

Fix has been submitted #2306

Cheredanine, 2017-02-08:
Both referenced are closed, however, helping a noob on uk quad site, new sprf3 flashed with bf 3.1.5, d4rii running ppm, dys sn16a (pwm) - no response from throttle, motors spun on arming but no change in speed on throttle stick. Reciever tab shows correct response from throttle is received.
Had him flash back to 3.0.1 and all good