betaflight/betaflight #2330: Furious FPV Moskito & FrSky, Betaflight connection issues

husafreak, 2017-02-05:
I have a Betaflight connection issue with my Furious FPV Moskito. Note that the quad is flying fine. If I connect the Moskito to my Mac with USB only I can see the quad and make setting adjustments and view screens normally. But if I power up the quad via its flight battery it will disconnect, or refuse to connect, to Betaflight. Occaisionally I saw a few seconds of connection before the returns from the quad freeze. So I am unable to do a transmitter setup in the Receiver tab. Thank you.

andymac4182, 2017-02-05:
Have you contacted Furious about this?

husafreak, 2017-02-06:
Yes, no response yet. I joined and posted on Bithub in case this is a common problem.

mikeller, 2017-02-06:
Question: Does this quad have an OSD? If so the problem is possibly that the OSD gets powered up when a battery is connected and then claims the MSP connection.

husafreak, 2017-02-06:
No OSD

mikeller, 2017-02-06:
Yeah, sounds like faulty hardware then. Just in order to better diagnose this, can you open a console (left click into configurator window, select 'Inspect'), and copy / paste the text that's printed in the window that is opening when the error occurs?

husafreak, 2017-02-06:
I am a mac user and couldn't figure out how to open a "console" on that. My PC is very old and wouldn't connect at the port. I'll try again later.While playing with this I actually got it connected to Betaflight with the receiver powered. I tried to do my transmitter subtrims and endpoints but afer about one minute it froze up again.So yeah, the board definitely has the capability to connect and will occasionally connect for a minute but then it freezes. After it freezes I have to reboot Betaflight before it will connect again with the receiver unpowered.

mikeller, 2017-02-07:
On a Mac you should be able to open the Inspector / Console window in the Chrome menubar under Tools / Developer tools.

husafreak, 2017-02-07:
Thanks, I did find the console under developer. I can't see it when Betaflight is selected. I have to go back and forth between Betaflight and Google desktop. So after the problem displayed itself I looked at the console but there is nothing there. There is no information there.I decided to try this on my big desktop Mac. On that computer the quad stayed connected to Betaflight much longer before freezing. I was able to do my centers and endpoints in the receiver tab. I noted that the Aux 5 channel is going crazy doing its own thing all the time. Now the Moskito only uses 8 channels and the FrSky Tx is set to D8 so Aux 5 on should not be registering anything from the Transmitter. Certainly the 9ht channel should not be wandering all over... channels 10 on are all at 1500. I will mention this to Furious FPV.I don't know if reflashing the board is a good idea or not. I appreciate your help, sorry I am not more computer savvy, plus my computers are both pretty old.

mikeller, 2017-02-07:
Ok, sounds like that gives you a workable solution then to allow you to do your Rx setup, and just connect without a battery on for the rest of the setup.
It sounds like in your setup the Rx uses Aux5 to send the RSSI value to the flight controller (try moving your Tx relative to your Rx and see how Aux5 reacts). You can use this in Betaflight by setting 'RSSI Channel' to 9.

husafreak, 2017-02-07:
Cool, thanks!

closed 2017-10-23 by jflyper