betaflight/betaflight #2333: Very IMPORTANT New feature request for both Betaflight and Cleanflight (ESC tab)

silentwindblows, 2017-02-06:
BLhelisuite can be problematic and very annoying, so we need a new ESC tab in Cleanflight and Betaflight, not just to configure and flash ESC (simonk, BLheli etc) firmwawre, but also to flash simonk and BLheli bootloader onto ESC's that do not come with a bootloader. The ESC tab must be comprehensive, extremely powerful and flexible, user friendly

andymac4182, 2017-02-06:
Have you tried the BLHeli Chrome Configurator?

silentwindblows, 2017-02-06:
I didn't know such a tool existed. Ate you telling me there's actually a
configurator specifically for BLhehi, that runs as a Chrome app ?

andymac4182, 2017-02-06:
That is correct. :)

andymac4182, 2017-02-06:
https://github.com/blheli-configurator/blheli-configurator/

https://chrome.google.com/webstore/detail/blheli-configurator/mejfjggmbnocnfibbibmoogocnjbcjnk

silentwindblows, 2017-02-06:
I just installed blheli in chrome, went back into Cleanflight configurator
to disable UART 3 MSP, which is normally used for CC3D boards flex port.
Then i successfully connected to the board in bLheli chrome, but every time
i click read setup, the log window just shows "reading setup", then "read
setup finished". Nothing is displayed after this. All VCP Drivers are
installed, latest cleanflight firmware on the CC3D flight Controller board,
so why isn't BLheli chrome not displaying the ESC settings after reading
reading setup, blank interface, just a little note top left of screen
regarding moving propellers and power the ESC's. Very strange. I've tried
all 3 ports on the CC3D (flex, main, USB). Green light is solid when i
connect to CC3D within BLheli chrome, so obviously bootloader mode seems to
be fine, but it isnt showing ESC settings

andymac4182, 2017-02-06:
You would be best to post this in the blheli-configurator repo.

silentwindblows, 2017-02-06:
Cheers

borisbstyle, 2017-02-06:
closing issue as the separate app already exists

closed 2017-02-06 by borisbstyle

fncisco, 2017-03-20:
You need power (battery) your esc before click on "read setup"