betaflight/betaflight #2335: IRC Tramp Channel 7 shown as F8

borisbstyle, 2017-02-06:
Channel R7 and F8 are both 5880Mhz

When choosing R7 the display shows F8. This is due to lookup list, but maybe there is a way to workaround it? What do you think @jflyper?

jflyper, 2017-02-06:
Let me look into it.

raphaelcoeffic, 2017-02-06:
@borisbstyle @jflyper the Tramp does not have anything like channels. You can only set frequencies. So starting with a fresh power cycle, we fetch the current frequency. Now when 5880MHz is set, what should we consider it to be? F8 or R7? There is no good solution to that... apart from getting rid of bands/channels....

jflyper, 2017-02-06:
We can remember the band/channel at the vtxDevice_t level.

raphaelcoeffic, 2017-02-06:
Anyway, that would only work if it's in EEPROM. Otherwise, you will never know on power up.

raphaelcoeffic, 2017-02-06:
Please note that LUA takes it's info directly from vtx_common when reloading the data after a change.

raphaelcoeffic, 2017-02-06:
Ok, so I could come up with a "fix" when changing channels, but not fixing the situation on power-up without saving anything in EEPROM. I think that would seem odd for users in case we fix it for changes made online, but not when powering up.

@jflyper I think CMS uses its own variables that won't be changed after channel change (at least in the menus), but the status line will say "F8" instead of "R7". With LUA/MSP, the menus will change as well.

jflyper, 2017-02-06:
I think we need a EEPROM storage for tramp. We need some for direct attached RTC6705 anyway.

raphaelcoeffic, 2017-02-06:
@jflyper ;-) Now we're getting there! Should I just do it? You can review afterwards.

closed 2017-02-08 by borisbstyle