betaflight/betaflight #2337: PIKO BLX accelerometer fails

halninekilo, 2017-02-06:
Bought an Eachine Chaser 88 with a PIKO BLX clone - dumped the original settings, flashed the board to 3.1 and got the 'acc missing' code.

So I ordered an original PIKO BLX, flashed it with 3.1.3, flew it - changed the PID's, disconnected it from the configurator - 'acc missing'.

Will try the third FC tomorrow... :(

halninekilo, 2017-02-06:
Looks like this isn't a firmware issue, just really crappy board design... 
Sorry to have bothered everyone :(

closed 2017-02-06 by halninekilo

mikeller, 2017-02-07:
@halninekilo: Just out of interest, would you mind sharing with us what the problem with the board design is?

halninekilo, 2017-02-07:
I don't feel qualified to comment on that in a meaningful way. I'm a software guy and know little about actual PCB design. 

Shouldn't have made that statement in the first place, but I was quite miffed - have over a dozen of FC's and none ever failed on me like that, let alone two within such a short time...

So take this with a grain of salt but from what I gathered from the web, the BEC is not up to snuff (&specs) and a chip, which I think is part of the voltage regulation, likes to fail. I assume this is what happened to me. 

The eachine clone is completely bricked while the original one actually manages to power the gyro/acc MPU after 10-30 tries...

The quad came with a 200mw VTX/cam combo and I have a hunch that this was too much for the 2A rated 'heavy duty' BEC.

mikeller, 2017-02-07:
Thanks. For whatever it is worth, there are other FCs that have had this problems, and for some it was addressed in a V2 with better BEC. Also, this shounds like a shortcoming that you should be able to address in a relatively easy fashion by replacing the BEC with a better off-board one like this: https://www.pololu.com/product/2850

halninekilo, 2017-02-07:
Thank you for the suggestion! Will look into other pico-sized FC's next, adding a secondary BEC just doesn't feel right for a 80mm nano ;)

halninekilo, 2017-02-07:
For protocol - received the third PIKO BLX today and so far it's working flawlessly.

I am now fairly sure that I was sold another clone before :( Both differ slightly.
So... hanging my head in shame. No offense @FFPV.

unitware, 2017-04-20:
"acc missing" is that the alternating green leds?