betaflight/betaflight #2339: No throttle response from 1500-1990. 2000 = full throttle

TempleClause, 2017-02-06:
There is something really wrong with version 3.1.3. 

From 1000-1500 everything is normal. But from there on there is almost no change in power until you reach 2000. Then the quad will full throttle. 

I have a blackbox to confirm this issue. It's in the 4th session and at 3:00 I start to slowly ramp up the throttle. But as you can see it "hangs" at 1500 and then suddenly skips to 2000. 
[LOG07171.TXT](https://github.com/betaflight/betaflight/files/755139/LOG07171.TXT)

There already posts on facebook with people that have the same issue:
https://www.facebook.com/groups/betaflight/permalink/415844085432017/

ScottWv10, 2017-02-06:
Yh same problem here, 3.1.3 i had this problem but soon as i went back to 3.1.1 all works fine.

borisbstyle, 2017-02-06:
Remove your throttle expo

TempleClause, 2017-02-06:
Then there is something wrong with the new throttle expo, I always used throttle expo. 
Downgraded to 3.0.1 with the same throttle expo and everything is normal again.

borisbstyle, 2017-02-06:
See release notes for it. There is a note about it that the old multiwii throttle expo is discontinued as it was causing another issue where min_throttle setting would affect your lowest throttle stick input

You nlcan also downgrade to 3.1.0 btw

TempleClause, 2017-02-06:
Just saw the release note, thanks Boris! 

So what would you suggest for getting the same throttle expo back again? Downgrading sounds like a bad idea because like you write it the throttle expo will stay the way it is?

ScottWv10, 2017-02-06:
yh im in same boat here, i would like to keep my throttle expo on newer version, as said above dont really want to downgrade for this.

Eganwp, 2017-02-06:
Put a throttle expo curve in your radio instead if you really need it. There will be no issues doing so.

borisbstyle, 2017-02-06:
The current throttle expo is really temporarily solution before custom throttle curve is implemented. It is not possible to get the same result as before.

3.1.0 is the best candidate where it still works if you dont want to wait till next release.

TempleClause, 2017-02-06:
@Eganwp That sounds like a really bad idea. That's one reason I hated KISS for ;-) 

@borisbstyle Ahhh okay so the throttle curve will not stay the way it is, that's what I wanted to hear :-) So I will downgrade to 3.1.0 and wait for the next release. Cheers buddy!

closed 2017-02-06 by TempleClause

borisbstyle, 2017-02-07:
Throttle expo restored in 3.1.4
https://github.com/betaflight/betaflight/releases/tag/v3.1.4