betaflight/betaflight #2340: XRACER_SPI blackbox download partial transfer

dmccue, 2017-02-06:
XRACER_SPI has built-in blackbox logging

Betaflight 3.1.3
Settings: http://pastebin.com/raw/1AeJwkdq

Symptoms: Erase betaflight logs, generate 1-2 mins of logs, plugin copter via USB, attempt to download logs, transfer hangs around 10% and never completes

Eganwp, 2017-02-06:
It's caused by CRC errors in MSP. It's a known issue and has been patched in #420.

dmccue, 2017-02-06:
we're on release 1012 currently so this should have been fixed by now?

dmccue, 2017-02-06:
Solution looks to be in https://github.com/betaflight/betaflight-configurator/pull/418
Issue with configurator, not betaflight
https://github.com/betaflight/betaflight-configurator/issues/411

AndersHoglund, 2017-02-06:
There is still a root cause for CRC errors to be investigated. Seems only to occur with Xracer_SPI boards, or?
The BFC update is however a better way of handling the CRC errors, retrying instead of failing. But is is not a correction for the real problem.

mikeller, 2017-02-06:
@AndersHoglund Yes, there definitely is a root cause. I think it's possible that the Xracer_SPI boards have some extra circuitry on the CP210x setup that introduces signal distortion. I have been using dataflash downloads with 4096 byte block size, and speeds up to 1000000 bps on a NAZE with CP210x for the better part of a year now, and never had a problem with CRC errors.

AndersHoglund, 2017-02-06:
I have an early XracerF3, non-SPI. Works just fine, too. I have however noticed Flash verification errors on high baudrates, so you may be correct there is something with the CP210x. No problems with BB downloads though, but I have not tested that on higher baudrates than 115200. 
Another suspect would be the data flash SPI connection, shared with gyro. On my boards the data flash is solo on the SPI. There was also an earlier problem with SPI back-to-back chip select signaling, #1779 and #1819. Could this still be haunting some boards? 
Well, not really an hi prio problem now with the BFC download retry change. But still annoying...

mikeller, 2017-02-07:
The SPI data flash aspect is an interesting angle. This could probably be looked at (and even mediated to some degree) by checking for the occurrence of download errors for different loop time  settings. (And if verified, at least we know for sure it's a hardware limitation and not a software bug.) By 'flash verification' you mean when flashing firmware, or dataflash download?

AndersHoglund, 2017-02-07:
I meant the verification phase when flashing firmware. 
How did you test BB download at 1000000baud ? Tried it, and my Xracer will not even connect. It connects only at 115200.

mikeller, 2017-02-07:
> I meant the verification phase when flashing firmware.

Ok, that makes it sound like it's the CP210x circuitry that's got something wrong on this board. I've never had a firmware flash verification error on a NAZE, and I've used one of them for testing for a while, so it got flashed a lot.

With CP210x based boards, in order to use different speeds to 115200, you need to first switch the speed of the MSP port in the ports tab, then reboot, and then switch the speed in configurator. Repeat every time the config is reset. (This is why I gave up using non-default speeds.)

AndersHoglund, 2017-02-07:
Ahh, of course. MSP first. No auto baud... 
But the Xracer did not connect at 100000baud anyway, and neither did my Naze. 
At 500000 Naze connected  but not the Xracer. At 250000 both connects. There is some difference....

So, I did a BB download from the Xracer, no problem at this speed.

    Connecting to: COM3
    serial.js:116 SERIAL: Connection opened with ID: 486, Baud: 250000
    three.min.js:536 THREE.WebGLRenderer 72
    onboard_logging.js:442 Dataflash dump file path: E:\Users\andho\Documents\Modellflyg\cleanflight\Backups\XracerF303V1-exp\bb-log\BLACKBOX_LOG_20170207_173311.BFL
    onboard_logging.js:332 Received 4558848 bytes in 226.63s (19.64kB / s) with block size 4096.

Not sure if this proves anything really.....

dmccue, 2017-02-07:
Looks like there's more than meets the eye, reopening

mikeller, 2017-02-07:
@AndersHoglund It really looks like the maximum feasible speed for CP210x based boards is dependent on the individual hardware config (my NAZE rev6 working at 1000000, yours not). This is the reason why we did not increase the default MSP speed in the end.

@dmccue: Having to adjust the speed on both sides of the connection is how serial connections work. VCP boards just hide that fact as they don't use a physical serial connection anywhere, and the baud rate setting in configurator is ignored in this case. Do you have a suggestion on how you would connect the flight controller's SPI port to the PC?

dmccue, 2017-02-07:
Tried on build #420 of betaflight-configurator and the retry seems to work well
https://github.com/betaflight/betaflight-configurator/pull/420

closed 2017-02-07 by dmccue