betaflight/betaflight #2341: Fury F4 Dshot conflit - Motor 3 with LED

jnrabreu, 2017-02-06:
Hi!

If i enable the LED on the Fury F4, Motor 3 dosent work. 

The problem is the DMA Stream.

With LED Strip OFF:
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

With LED Strip ON
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Any soluction ?

jnrabreu, 2017-02-06:
Note: Othe protocols (Oneshot; Multishot) Work fine

mikeller, 2017-02-07:
Just off the top of my head, without further investigation, you could try to see if moving LED to pin `C09` (the PPM in pin) solves this (i.e. DMA assignment for MOTOR 3 is showing).

borisbstyle, 2017-02-07:
@jnrabreu I suggest trying different timers as referenced in STM32 manuals. let us know if you find solution for this target.
None of us has it so we cant troubleshoot on it.

jnrabreu, 2017-02-07:
Hi!

Can you give me some pointers, on how to do that ?

mikeller, 2017-02-07:
@jnrabreu Have you tried remapping LED to the PPM pin?

jnrabreu, 2017-02-07:
Yes, i did. It's works like that. But would be nice to work on it's own pin.

mikeller, 2017-02-07:
Good to know, at least that provides us with a last resort. To check if you can find a timer / dma config that works with the proper pins you'll have to build your own firmware, and adjust the timer / channel assignments in `src/main/target/FURYF4/target.c`. Keep in mind that they will have to match the available timers / channels per pin as defined in http://www.st.com/content/ccc/resource/technical/document/datasheet/ef/92/76/6d/bb/c2/4f/f7/DM00037051.pdf/files/DM00037051.pdf/jcr:content/translations/en.DM00037051.pdf pp 47.
The DMA streams used by the individual timers can be looked up in http://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf pp 309.

If you are not set up to do the builds yourself, I can try figuring out a working configuration for you, but I can't test since I don't have a board, so I'll have to work with you.

mikeller, 2017-02-07:
In giving `src/main/target/FURYF4/target.c` a quick look, just replacing the line

```
    DEF_TIM(TIM5, CH1, PA0, TIM_USE_LED,   TIMER_OUTPUT_ENABLED, 0 ), // LED_STRIP - DMA1_ST2_CH6
```

with

```
    DEF_TIM(TIM2, CH1, PA0, TIM_USE_LED,   TIMER_OUTPUT_ENABLED, 0 ), // LED_STRIP - DMA1_ST5_CH3
```

might be a winner. Please let us know if it works.

blckmn, 2017-02-07:
I have one of these i think so may be able to test tonight if we don't hear back.

kc10kevin, 2017-04-22:
Sorry guys, been on an extended break from the scene.   Just found this issue raised several months ago.  

Couple of options availble for this.  @mikeller option above (not sure if @blckmn had time to test it out) or changing the motor 3 timer to:

`DEF_TIM(TIM8, CH3N, PB1, TIM_USE_MOTOR,                     TIMER_OUTPUT_INVERTED, 1), // M3 - DMA2_ST4`

or

`DEF_TIM(TIM1, CH3N, PB1, TIM_USE_MOTOR,                     TIMER_OUTPUT_INVERTED, 0), // M3 - DMA2_ST6`

Give me a couple days to get back from vacation and get the scope out to test.

kc10kevin, 2017-04-29:
I ended changing the motor 3 timer (PR #2978).  I tried changing the LED strip timer to Tim2, Ch1, but it had a strange effect on motor 1 & 4 with the LED strip enabled (See Video).  With LED Strip disabled, all protocols look fine.  Did not have time to test with motors to see effect. 

https://www.youtube.com/watch?v=66XCF2SSc60

Ended up using the following and everything looks good on the scope now:

`DEF_TIM(TIM8, CH3N, PB1, TIM_USE_MOTOR, TIMER_OUTPUT_INVERTED, 0), // M3 - DMA2_ST2`

@jnrabreu - target can be downloaded here for you to test:  https://github.com/kc10kevin/betaflight/releases/tag/v3.2

closed 2017-04-29 by kc10kevin

mikeller, 2017-04-29:
@kc10kevin:

> I ended changing the motor 3 timer (PR #2978). I tried changing the LED strip timer to Tim2, Ch1, but it had a strange effect on motor 1 & 4 with the LED strip enabled (See Video). With LED Strip disabled, all protocols look fine. Did not have time to test with motors to see effect.

Not so strange really. One timer can only be shared between things that require identical timing (i.e. only motors _or_ servos). Sharing TIM2 between motors and LED won't work.

kc10kevin, 2017-04-30:
@mikeller:

Got it!  Thanks!  Learn something new every day.   I'm surprised there are not more complaints then.  There are several targets that share timers between LED and Motors.  
AIRHEROF3
ALIENFLIGHTF1
ALIENFLIGHTF4
BEEROTORF4
BLUEJAYF4
KAKUTEF4
KIWIF4

etc....

Are these targets motors (shared) not working when the LED Strip is enabled?  I have not tested with motors, just on the scope.

mikeller, 2017-04-30:
Yes, this is the case. Without checking all of these targets, I am assuming that the sharing happens for motors > 4. It is kind of accepted that for craft that use more than 4 motors, compromises with respect to other features are required. For some of these boards (like the BLUEJAYF4), the output of the LED_STRIP is sent to the motor 5 pin anyway, so there is no way to use more than 4 motors _and_ the LED_STRIP without the remapping of resources (which remaps timers associated with pins as well).