betaflight/betaflight #2342: [3.1.3] get telemetry_inversion  => Invalid name

MiroRadenovic, 2017-02-06:
Hello ,
i'm trying to set up telemetry and I always receive Invalid name

```
# version
# Betaflight / SPRACINGF3 3.1.3 Feb  2 2017 / 13:58:33 (73b498e)

# feature TELEMETRY
Enabled TELEMETRY

# get telemetry_inversion
Invalid name

# set telemetry_inversion = ON
Invalid name
```

thank you,
Miro

wind0r, 2017-02-06:
it got renamed to tlm_inversion.
sorry for the inconvenience!

mikeller, 2017-02-06:
Hi @miroradenovic. `telemetry_inversion` has been renamed, it's `tlm_inversion` now.

MiroRadenovic, 2017-02-07:
sorry guys, but this still looks like it's in work in progress

```
# version
# Betaflight / SPRACINGF3 3.1.3 Feb  2 2017 / 13:58:33 (73b498e)

# get tlm_inversion
tlm_inversion = ON
Allowed values: OFF, ON

# set tlm_inversion OFF
Invalid name

# set tlm_inversion 0
Invalid name

# 
```

mikeller, 2017-02-07:
@MiroRadenovic: The correct syntax is `set tlm_inversion = off`.

MiroRadenovic, 2017-02-07:
thx, it works!
thank you guys for the great work with betaflight!

closed 2017-02-07 by MiroRadenovic