betaflight/betaflight #2343: Arming and no respond

2black0, 2017-02-06:
Hi, I have airbot F3 Board (spracingf3) + Radiolink R9D (sbus) + ESC 12A (Pwm based)

Sbus over uart2 and it run normally, I can get respon of stick and switch over receiver tab,

I can make test motor too, run from minimum and maximmun without problem

But something weird happen, when arming (in horizon/acro mode) motor spin in minimum throttle but movement stick dont make effect, I move throttle, pitch, roll, yaw stick to max and motor still in minimum throttle (1070)

I check in receiver tab when armed, everything just normal, receiver work normally

Flashed with 3.1.0 and 3.1.3
I was build many quad based on sprcing f3 board, and no problem like this, so anyone can make suggestion for me what should I check or I do?

Noctaro, 2017-02-06:
Is this the same problem? -> https://github.com/betaflight/betaflight/issues/2091

2black0, 2017-02-06:
update:
Ok, i tested your suggestion.
Gyro update at 2.67KHz
PID loop at .53KHz
I can confirm this works. Was able to do a normal flight with and without ACC enabled.
Switched back to the old PID/Gyro (2k/0.5k) loop settings and again wasnt able to take off. (Also tried to disable ACC, seems to make no difference)

With 1k/0.25 it also is flyable.
4k/0.5k ,4k/0.44 (CPU 16%) are not flyable

Its solved? @noctaro

2black0, 2017-02-06:
ok thanks, solved, will maiden today

closed 2017-02-06 by 2black0