betaflight/betaflight #2344: A default custom motor mixer will show up in the diff even if not modified

MJ666, 2017-02-07:
@martinbudden all AlienFlight controllers have a build in custom motor mixer defined. This is used for clean wiring on the Octo boards. I understand today this also can be done with an simple remap of the motor pins. Since the PG changes this custom motor mixer will always show up in the CLI diff output even it is not change and part of the default configuration. I have already looked into the code but have not found the root of the behavior yet. Likely the source of the problem is somewhere in this function:

https://github.com/betaflight/betaflight/blob/master/src/main/fc/cli.c#L2004

martinbudden, 2017-02-07:
@MJ666 , PG changes are still a work in progress. I'm aware of this issue, but haven't addressed it yet in betaflight.

It should, however, be working in iNav. Can you test the iNav code to see if the problem exists there?

MJ666, 2017-02-08:
I checked INAV and actually the target specific custom parameters are not enabled and not working for the AlienFlight F3 target. After activating them all custom parameter show up in the diff output. so likely the are not accepted as part of the default configuration like in BetaFlight. I never touched the AlienFlight code in INAV since the real existing F3 boards are not relay useful for navigation. The don't have the required sensors or serial ports exposed to connect an GPS.

martinbudden, 2017-02-08:
That's interesting to know. So some work is required there. I'm not sure how AlienFlight got on the iNav target list. Perhaps since there are things not working and it is not really suitable for iNav it should be removed as an iNav target.

MJ666, 2017-04-26:
@martinbudden the issue is still present in latest master. The AlienFlight custom mixer definitions show as modified in the diff CLI output.

BTW: also did some updates for INAV and the problem is the same there.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis