betaflight/betaflight #2346: Motor spin after arming but don't react to Control

Olgidos, 2017-02-07:
Hello
I just tried to confid my new spracingf3 with beta flight. Everything worked so far, but now I can't get my motors spinning faster then slow^^

In the motor tab, when I give them power (2000) then they spin like crazy, so it's nothing fals connected.
In receiver tab my throttle goes from 960 to 2010.

But after arming, they will spin with some throttle but there isn't any difference between min throttle and max. 
I am a bit confused =/

Motors are spinning from weak to crazy in the motor tab -> so they are well calibrated 
Receiver is recognized-> slider works in receiver tab... (frsky subs)
Esc are in pwm Mode 
Maybe someone knows help =)

Gaz-B, 2017-02-07:
I have exactly the same issue on the builds just released try flashing 3.1.3 and make sure that stops the issue for now .

TempleClause, 2017-02-07:
3.1.4 same here, 3.1.3 works fine

ScottWv10, 2017-02-07:
same here for 3.1.4, strange issue haha

Olgidos, 2017-02-07:
Hmm i just flashed 3.1.0 an still got this problem

ScottWv10, 2017-02-07:
i flashed back to 3.1.1 and all works again

TempleClause, 2017-02-07:
For me Roll, Pitch, Yaw will still spin up the motors but throttle wont...

Gaz-B, 2017-02-07:
That is odd because i tried it again fresh download and it's exactly the same plenty of control authority but no throttle.

Olgidos, 2017-02-07:
I don't know if this is normal but I now get mor throttle when I push the stick.
Put sometimes there is enough power and sometimes it goes again down a little bit so I can't take off is this normal in horizon mode?
And in airmode, the throttle goes only up slowly

Olgidos, 2017-02-07:
I only get it in the air if I push jaw pitch or something

Gaz-B, 2017-02-07:
Olgidos What ever is wrong is still wrong I would stay grounded for now or revert to an earlier release in horizon mode if accelerometer is calibrated it should take of level using only throttle.

mikeller, 2017-02-07:
It's a bug in 3.1.4. 3.1.5 fixing it will be released soon.

TempleClause, 2017-02-07:
Sweet 3.1.5 fixed it for me 👍

closed 2017-02-08 by borisbstyle