betaflight/betaflight #2352: motor tab not working correctly with Dshot and 3d mode

Dracen-Knights, 2017-02-08:
ok so I am setting up my first Dshot quad and after many hours of test to figure out why the motors tab is not reflecting the fact I'm in 3d mode I found out that it's a glitch. I also can not get the passthrough to work when in 3d mode. now if I set it to oneshot 125 or any other non dshot  the motors tab reflects 3d mode and I can use blheli suites to check the esc's .  once any dshot is set then the motors tab doesn't work and passthrough doesn't work either.

My setup is as fallows

Armattan Chameleon Frame
xracer f303 v3.1 Running BF 3.1.3
Armattan 30a esc's (dshot compatible) 
Armattan Oomph motors
Armattan 5 & 12v regulated PDB
Runcam owl plus camera
Eachine 600mw VTX
Furious Piggy OSD

I was really hoping to be able to use the motors tab to test the setup but at this point I can't. So how do I set deadband? or low and high points?

mikeller, 2017-02-08:
Hi @Dracen-Knights. I think I know what the problem is (DShot settings overriding 3D settings in the motors tab), it stems from DShot support being added to the motors tab before 3D DShot was implemented. I will look into fixing this. This is more of a betaflight-configurator issue, so can you please open an issue on https://github.com/betaflight/betaflight-configurator/issues/ ?

closed 2017-02-08 by mikeller

Dracen-Knights, 2017-02-08:
what part ? all of it?
  I'll simply add a copy of this over there as well. I can edit it there if need be as well.