betaflight/betaflight #2354: Limited AUX channel with 3.1.5

alexjx, 2017-02-08:
Updated to 3.1.5, the max_aux_channels are now limited to max 6. This actually limited the potential channels to 10. For Frysky XM and XM+, where the RSSI is mapped to channel 16, that betaflight could not use that channel anymore.

mikeller, 2017-02-08:
Just set `max_aux_channels` to 13 (or however many AUX channels your setup supports).

alexjx, 2017-02-08:
Before 3.1.5, I could set max_aux_channels to 12, but now it's cap at 6.

jflyper, 2017-02-08:
`MAX_AUX_CHANNELS`, 20 for F4 and F7 (STM_FAST_TARGET), and 6 for F1 and F3 is used in cli.c.

I can't find a commit that cli.c was changed to use `MAX_AUX_CHANNELS`.

mikeller, 2017-02-08:
@jflyper: https://github.com/betaflight/betaflight/commit/1ec8adbb5c4dff9166f83e875706ff59c958bdf2

jflyper, 2017-02-08:
I guess @borisbstyle have over fixed the max aux out of range issue?

borisbstyle, 2017-02-08:
@mikeller 
See comments on cli.c
https://github.com/betaflight/betaflight/pull/2338#pullrequestreview-20402296

> mikeller 2 days ago Member
> Should use MAX_AUX_CHANNELS here.

borisbstyle, 2017-02-08:
@jflyper No I fixed the out of the range default, but @mikeller suggested to not use 20, but to use the defines in common.c
I assumed those were based on something.

jflyper, 2017-02-08:
@mikeller So, max aux = 6 for F1 & F3 is a feature?

mikeller, 2017-02-08:
@jflyper Seems to be. I didn't introduce them, my comment was because of the use of a magic number. Addressed in #2359.

SpoonFPV, 2017-02-09:
There needs to be a setting for RSSI AUX channel, or allow max aux channels to be up to 12 to use the micro FrSky receiver RSSI function.  It's output on AUX 12 ... channel 16 so need to be able to set max AUX to 12 to use the feature.

borisbstyle, 2017-02-09:
I scheduled a fix for 1st of March in 3.1.6 maintanence release

closed 2017-02-21 by blckmn