betaflight/betaflight #2364: CC3D betafligh 3.1.3 Arm issue

Paulatkinson01, 2017-02-09:
When I turn my radio on first and then power the my quad, the quad will not arm. If I turn my quad on first and then radio, it will arm. If I had the CC3D connected to my computer is would always arm.

I had my i-bus receiver plugged into the flexi port, switching to the Main port for i-bus and the issue goes away.

mikeller, 2017-02-10:
This sounds very much like an issue with your RX, it seems to not like being powered on while the TX is already sending a signal. Have you tried contacting the RX' manufacturer about this?

Paulatkinson01, 2017-02-10:
They wouldn't explain why it works perfectly in the other port. Also when the issue occurs the ESC don't finish the startup tones.

mikeller, 2017-02-10:
Ok. The ESCs not initialising properly makes it sound like the flight controller is not starting up properly. I can't do much more investigating into this because I don't have a board to test with. But as you say, it works fine if you connect the RX to the main port, so you probably just should do that.

Paulatkinson01, 2017-02-10:
Is there anything I can do to investigate this further?

atomiclama, 2017-02-10:
I can look into this next week if no one has sorted you out by then.
Which RX are you using?
I'm on IA6B with CC3D not sure of the version but it's fine.

If you get it into the state where it will not arm can you then plug it into the computer and see if the RX channels are working in the configurator?

Paulatkinson01, 2017-02-10:
Hi Atomiclama,

Same receiver with an i10 transmitter, I think if I remember correctly I don't think I could connect to the flight controller over USB when this happens (i'll check when I get home) more than happy to help where I can, just not sure where to start.

Cheers,

Paul01 (MRUK) :-)

atomiclama, 2017-02-15:
Ok verified this problem on 3.1.3 and 3.1.5

This is on the bench with a known working TX/RX.

`
CLI

# diff
# Betaflight / CC3D 3.1.5 Feb  7 2017 / 22:04:38 (4646f9d)

feature -RX_PPM
feature RX_SERIAL
serial 2 64 115200 57600 0 115200
aux 0 0 0 1600 2075
set serialrx_provider = IBUS
profile 0

rateprofile 0
rateprofile 0

#
`

The problem is that with the TX turned on first the RX binds and starts outputing data before the FC has booted up properly. 
This is preventing the FC from completing it's boot up. The FC locks up no USB comms, nowt.
 I have not found where it is getting to yet.
 If the FC boots up fully before the TX is turned on and the RX binds all is ok.

It may have something to do with interrupts not being handled properly during startup on the flex port. The main port is ok.

mikeller, 2017-02-15:
@atomiclama: Could it be that the signal on the flex port is causing the MCU to go into DFU mode?

atomiclama, 2017-02-16:
@mikeller I'm not fully up to speed with DFU / bootmode / etc /etc. But I thought DFU was for the USB updateable F3 and up processors. Didn't think the F1 supported this. 

I'll get the debugger on it today and have a look.

mikeller, 2017-02-16:
@atomiclama: Yes, DFU is used for firmware update on all STM32 MCUs. It works on USB / VCP, and it works on some of the UARTs, which ones is dependent on the model.

AndersHoglund, 2017-02-16:
No, not entirely correct. DFU is a protocol used on USB only, and is not available on all devices. Of course we can do Firmware Update on all devices, but with different protocols on different Devices and depending on connection. Calling all firmware update procedures "DFU" is a bit incorrect. 

Specifically on CC3D, with a F1 chip, there is only the USART protocol on UART1 available, no DFU over USB. ( For flashing over USB you need the OPBL bootloader and GCS, but that is another matter).

See the ST app note, table on page 21 for an overview on DFU, USART and I2C protocols. 
www.st.com/resource/en/application_note/cd00167594.pdf

Details on USART here:
www.st.com/resource/en/application_note/cd00264342.pdf
Details on DFU here:
www.st.com/resource/en/application_note/cd00264379.pdf
Details on I2C here:
www.st.com/resource/en/application_note/dm00072315.pdf

atomiclama, 2017-02-16:
Ok it is getting stuck in the I2C error handler 

i2c_er_handler

The flexport shares with I2C. Looks like something is enabling the I2C for some time before the serial port takes over. Data from the RX is forcing it into error.

Can't give a call stack as my debugger has stopped allowing me to break execution. I'll see if I can get me debugger going this afternoon. Think a recent update of eclipse has screwed things :-(

atomiclama, 2017-02-16:
Why is 
#define USE_I2C
still defined ? As support for BARO and MAG has been removed, Can't see peripherals that would use the I2C bus. Could this be the problem? I2C enabled but nothing to handle it?

mikeller, 2017-02-16:
@atomiclama: Two more peripherals that use I2C are GPS and DISPLAY. Both have been removed for F1 targets, so I2C can probably be removed from CC3D.

mikeller, 2017-02-17:
@AndersHoglund: Good point. Fixed in #2423.

atomiclama, 2017-02-17:
removed I2C via #2416

commit 1bfec057 (2017-02-19): Merge pull request #2416 from atomiclama/bf_ibus

Fix #2364 remove defines for I2C
commit cc8c12d3 (2017-02-19): Fix #2364

USE_I2C was still defined even though no I2C devices BARO MAG were
defined. This led to the I2C being active but not being handled. So any
data on the I2C line caused it to error and not be handled.

Tested with IA6B RX hooked upto the flexport of the CC3D with TX on
before the FC boots.
closed 2017-02-19 by martinbudden