betaflight/betaflight #2366: Naze 6 - servo_tilt not forwarding channels on motor outputs above version 1.9.0

TomasKot, 2017-02-09:
I have flashed today with release 3.1.5 and I'm not able to spin the motors 1,2....just spinning motors 3,4 on 1, 2 position. The pins on motor output are not working so I was flying with the servo_tilt feature on and soldered the motor1 to pin3, motor2 to pin4, motor3 to pin5, motor4 to pin6 and in version release 3.1.5. just the motors 3 and 4 sniping on the position of motor1 and 2.

How can I male it working?

mikeller, 2017-02-10:
I am assuming what you are trying to do is to connect the servos for SERVO_TILT to the pins for motor 1 and 2?

If so, did you do the pin remapping to assign the correct functions to the pins:

```
resource servo 1 a08
resource servo 2 a11
resource motor 1 b06
resource motor 2 b07
resource motor 3 b08
resource motor 4 b09
```

TomasKot, 2017-02-10:
This is exactly what I was looking for and was unable to do. I was using wrong remapping pins. Now working as it should. Do you know where I can find some datasheet of the output pins what can be used?

mikeller, 2017-02-10:
Unfortunately there is no easy way to do this at the moment, as we are looking into ways to make this more user friendly. If you do `resource`, you can see the default pin assignments for a quadcopter (only motors 1 - 4 used). Starting from there, you can assign the pins for motors 1 - 2 to the servos, and assign the pins for motors 3 - 6 for motors 1 - 4. The reasons for why it has to be this way are even more complicated, but can be seen in the timer assignments to pins in `src/main/target/NAZE/target.c`.

jflyper, 2017-02-11:
@TomasKot 
It's described in the Wiki page [SERVO_TILT for 3.1](https://github.com/betaflight/betaflight/wiki/SERVO_TILT-for-3.1).

Generic pin availability is hard to find by general users as @mikeller said.
I wish target maintainers would come up with a list similar to the one like this:

#### Resource mapping (WIP)

| Label    | Pin | Timer | DMA | Note                             |
|----------|-----|-------|-----|----------------------------------|
| RSSI     | A0  |       |     | Not ADC capable                  |
| PWM5     | A1  | 5,2   | ?,? |                                  |
| PWM4     | A2  | 2,3   | 1,1 |                                  |
| PWM3     | A3  | 2,4   | 1,6 |                                  |
| PWM6     | A8  | 1,1   | ?,? |                                  |
| UART1 TX | A9  | 1,2   | ?,? |                                  |
| UART1 RX | A10 | 1,?   | ?,? |                                  |
| PWM1     | B0  | 3,3   | 1,7 |                                  |
| PWM2     | B1  | 3,4   | 1,2 |                                  |
| LED      | B6  | 4,1   | ?,? | J9 (Not accessible with v3.1)    |
| PPM      | B14 | 12,3  | ?,? |                                  |
| CH2      | B15 | 12,4  | ?,? |                                  |
| UART3 TX | B10 | 2,3(!)| ?,? |                                  |
| UART3 RX | B11 | 2,4(!)| ?,? |                                  |
| CRNT     | C1  | ---   | 2,4 |                                  |
| VBAT     | C2  | ---   | 2,4 |                                  |
| UART6 TX | C6  | ?,?   | ?,? |                                  |
| UART6 RX | C7  | ?,?   | ?,? |                                  |
| CH5      | C8  | 8,3   | ?,? |                                  |
| CH6      | C9  | 8,4   | ?,? |                                  |

ThomasH3, 2017-03-19:
Hi. I remapped b08 and b09 to servo 1 and servo2. After I saved the setup I had problems with the motors on  b06 and b07. They started shaking about having any throttle input. Servos also don't work on a08 and a11

mikeller, 2017-03-20:
@ThomasH3: Yes, that won't work. Use A8 / A11 (motor 1, 2) for servos, and remap motors 1, 2, 3, 4 to B6, B7, B8, B9.