betaflight/betaflight #2368: DSHOT Timer problem

KyunghyunLee, 2017-02-10:
I am using custom F4 FC with betaflight.
I made my own target and used it with no problem until 3.1.0

Recently, I modify 3.1.5 code for my target, I found DSHOT problem.
My FC is using Timer 1 channel 1-4 for output channel 1-4.
Other channel works fine, but channel 2 has problem. 

DSHOT frame is 16 bit, (18 byte for DMA), but in channel 2, 
number of frame is varying with channel 3 or 4 output value. 
When channel 3 or 4 value is 1000, channel 2 generate also 16 bit frame.
When channel 3 or 4 value is 2000, channel 2 generate only 7 bit frame.

And I found some targets which use 4 channels from 1 timer changed their target.c file for DSHOT.
It seems I cannot use all 4 channels from 1 timer. 

Is there any reason for this problem? How can I figure it out?

KyunghyunLee, 2017-02-10:
https://github.com/betaflight/betaflight/commit/b4060844ebda64c3c7275fdcfd098837c7efd27f
https://github.com/betaflight/betaflight/commit/13132e40bf2714ab5a52f73485cc24da190dfdd8

These two commits changed target.c file for DSHOT, 
They were also 4 channels from 1 timer, and changed some to other.

MJ666, 2017-02-11:
Each timer channel need to have an own uniqe DMA for DSHOT to work. Sometimes it's pasible the use an different timer for the same pin to make this happen. Also other functions like the ADC or SDCARD will conflict with the DMA channels. This conflicts also need to be resolved. Do this for an existing design is like a puzzle and is sometimes even not solveble. This is the reason why for some targets oter pins get used like remap the motor to the PPM pin or other motor pins.

KyunghyunLee, 2017-02-14:
There was a bug in my configuration.

closed 2017-02-14 by KyunghyunLee