betaflight/betaflight #2373: RotorRacer Racebase FC

nightlightup, 2017-02-10:
I keep having an Issue with this flight controller. The motor 2 does not want to work in motor tab. It does not spin. If I will go to Cli and exit then only Motor 2 will spin constantly at about 70%.
Have tried flashing 3.1.5, 3.13 and 3.0.1- same thing. I have no clue if it is the FC or software Issue. ( i presumed it is software since the M2 is spinning ) 
thanks.

mikeller, 2017-02-11:
A little bit more information would be helpful. Can you post the output of `diff` please?

nightlightup, 2017-02-11:
here's the diff:
# diff

# version
# BetaFlight/RACEBASE 3.0.1 Sep 25 2016 / 10:55:43 (2d83570)

# name

# mixer

# servo

# servo mix

# feature

# beeper

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# rxfail

# master
set osd_rssi_pos = 3094
set osd_gps_sats_pos = 1410

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

mikeller, 2017-02-11:
Have you tried calibrating your ESCs? It could be that the timing for ESC 2 is horribly off.

nightlightup, 2017-02-11:
yes have tried it several times..... and on diff versions of Betaflight... and nothing.... In motor tab the motor will not spin. If i shortly go to CLI and do nothing but just exit...it starts spinning like crazy...so I presume it is working just fine.... will try swapping the ESC to see if maybe it, is the cause of all this...although i highly doubt it...

nightlightup, 2017-02-11:
here's the dump info ..maybe it will say more than DIFF:
# dump

# version
# BetaFlight/RACEBASE 3.0.1 Sep 25 2016 / 10:55:43 (2d83570)

# name
name -

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature RSSI_ADC
feature OSD

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
[... 354 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

nightlightup, 2017-02-12:
Consider Issue as closed. Problem: ESC had a different BLheli Suite compared to the other 3. (Updated them to latest and it is ok, although timing is a bit off. Will change all esc for sure after a few flights)

closed 2017-02-12 by nightlightup