betaflight/betaflight #2375: BFF3 - OSD  disappears during flight

GaspR, 2017-02-10:
The OSD disappears during flight. I move the switch that on/off OSD but this unsuccessful.
https://youtu.be/-fHeVRNKfTY

I see an issue (#1568) on Omnibus F3 that closed due to non software issue.

My CF is Betaflight F3.
My VTx is on VBAT (tramp hv) and the camera is alimented by it. 
I only put the signal camera in Vin on CF and Vout to the VTx.
I don't know if it's hardware or software or an error from me, I may find the blackbox data if it's help.

My firmware is 3.1.5, I saw it before, but there was the bug on the on/off switch OSD, so I don't report it.

Eganwp, 2017-02-10:
This is almost certainly due to ESC noise blacking out the MAX chip and exasperated by damped light breaking causing large noise spikes and newer powerful motors. This same issue is a constant complaint on all OSD AIO boards (Omnibus, Matek HubOSD, some BFF3 boards, etc). 

There's a good thread on the DShot RCG thread about which combo of capacitors fixes the noise best. All mine have been resolved through use of caps as well.

jflyper, 2017-02-11:
@GaspR 

Read this and install ground wires CAM-FC for FC-VTX.
https://www.rcgroups.com/forums/showthread.php?2789895-OMNIBUS-Series-CAM-OSD-VTX-wiring

If it still doesn't go away, you can try the cap @Eganwp suggested.

And please direct your support questions on RCGroups thread. This is a place to raise and discuss software issue.

GaspR, 2017-02-11:
Thanks, that seem clear. Now i know is hardware and not soft, i close this here and go to rcgroups.
Maybe that will help someone to go at the right place.

closed 2017-02-11 by GaspR