betaflight/betaflight #2377: Cant choose SerialRX JETIEXBUS on 3.1

RoadyFPV, 2017-02-11:
Hi 
i have n Naze32 Board. In BF 3.0.1 i can aktivate RX Serial and JetiEXBus.
In BF 3.1.0 and later, i cant choose RXserial and JetiEXBus. It will be dont saved at restart.

Greetings Christoph

mikeller, 2017-02-11:
Hi @RoadyFPV.

Unfortunately, since there is not enough flash space available on F1 based boards, some features had to be removed to make space for the improvements in the core features. JetiEXBus is one of the features that has been removed from F1 based boards for 3.1.

You might be able to build your own binaries for NAZE with JetiEXBus included by dropping some other features you are not using, but the best long term solution is to upgrade to an F3 of F4 based board.

RoadyFPV, 2017-02-11:
Hi mikeller,

much thanks for your fast help. Ok, i cant build my own binaries so i'll use bf 3.0.1 for now, later i will upgrade my nighthawk 250 to f4 board.

Greetings

closed 2017-02-11 by RoadyFPV