betaflight/betaflight #2378: Dshot600 not working in 3D mode!

nori-modelar, 2017-02-11:
Using betaflight 3.1.0

In Dshot escs wont change direction. Tried old oneshot and everything works fine. 
So when i arm the quad and move the stick to the center it starts up like crazy. (actually its on half speed) Then if i go to full throttle its like a full throttle (ok) but when i go below center it goes into predicted speed value and when i try to go till the bottom, motors start up a bit and then slow down as i go to the bottom. So its just like a normal mode and there is no change of direction. ESCs are fine and it is working on other versions. (oneshot, multishot...)

mikeller, 2017-02-11:
What you describe sounds like your ESCs do act like 'bidirectional' is off when DShot is used. What type / brand of ESCs are you using?

nori-modelar, 2017-02-11:
maverick 24A  it is in 3d mode

mikeller, 2017-02-11:
Ok. This isn't a blheli_s ESC, so its configuration is different to how blheli_s ESCs work. From the sound of it, configuring it for 3D mode does ***not*** enable 3D mode for DShot. There is a way to enable 3D mode from within DShot, without the use of a configuration tool, but it's not yet implemented in Betaflight.

Sorry, but it looks like for the moment you'll have to use either DShot *or* 3D mode with this ESC.

borisbstyle, 2017-02-11:
Sounds like maverick esc doesnt support dshot 3d mode.

Please ask maverick how to activate dshot 3dmode on their esc

Gaz-B, 2017-02-12:
This sounds similar to what i am getting with Flycolor Raptor 30a BLHeli-S on 16.6. Using the Motor tab I get throttle raising from 0-50% at 50% it sounds flat out, then it repeats again from idle again 51-100%  at 100% once again it sounds flat out. not sure if it reverses at the 50% reset tried on D-Shot 150,300 and 600 Works perfectly on all non digital protocols. Betaflight 3.1.5

borisbstyle, 2017-02-12:
@Gaz-B Thats how 3D works on Dshot. It works differently than on normal PWM.
Just fly it and you will see it works.

3D on dshot has lower section of throttle reversed actually

Gaz-B, 2017-02-12:
So are you saying it will work from the TX but is reversed on the bottom half using the motors tab ?

borisbstyle, 2017-02-12:
Yes. Motor tab handles it all the same

Gaz-B, 2017-02-12:
I seem to have two ESCs that can't do dshot constantly rebooting and wont spool up other two seem fine  I have another later build running same escs been running D600 perfectly for ages. i will leave this one on Mshot for now.

mikeller, 2017-02-12:
@Gaz-B: Have you checked firmware versions on your ESCs? The symptoms could be caused by older firmware on the two ESCs that don't work properly. Or else it could be filter caps that are borderline ok for DShot 600, try DShot 150 to check.

Gaz-B, 2017-02-12:
these esc are now being listed as d600 compatible I tried 300 think i tried 150 but will confirm.

Gaz-B, 2017-02-12:
same on d150 two work One of the two that don't goes full speed after a few throttle pumps only way to stop it is to pull the battery. throttle pumping has no affect on this motor before it goes mad.

mikeller, 2017-02-12:
@Gaz-B: Have you checked firmware levels on them, and updated if required?

Gaz-B, 2017-02-12:
Tried 16.6 and 16.65 adding tunes didn't help.