betaflight/betaflight #2379: Eachine Falcon 250 Pro  SPracingF3 model

Brisingrox, 2017-02-11:
I don't know if this the right place to ask for help but i gone try.

I've bought a Eachine Falcon 250 pro SPracing F3 and it comes with cleanflight 1.12 out of the box. I did upgrade to Betaflight 3.1.5 and it was a mess. since that point i did search all over the web for a solution but i did not find one? is someone here who could give me some advice how i could pass under betaflight and make my Falcon 250 pro flyable?

Thanks

jflyper, 2017-02-11:
@Brisingrox This is a software issue reporting place.

There is a RCGroups thread where you may get some support.
[eachine Falcon 250 Pro - CC3D/F3, 2204 2300kv, 700tvl cmos, 200mw vtx - $220-227](https://www.rcgroups.com/forums/showthread.php?2704364-Eachine-Falcon-250-Pro-CC3D-F3-2204-2300kv-700tvl-cmos-200mw-vtx-220-227)

enc0de101, 2017-02-19:
add 90 degree to yaw offset

Brisingrox, 2017-02-19:
I have done that! I hear a read from more people who seems to have the same problem. Like sudden fulk power without having control over the quad.

blckmn, 2017-02-21:
What are some of the debug outputs. Like how much cpu is being used? What does the resource command say?

Have you hit reset settings and saved?

closed 2017-03-19 by Brisingrox