betaflight/betaflight #2380: New PID restriction by Selected ESC Protocol my have Unintended Consequences

spatzengr, 2017-02-11:
Check out the video at https://youtu.be/Q2nXUQku-D4 in regard to how restricting the PID loop to what can be transmitted by the ESC protocol can have some unintended consequences (it seems?).  With this, I don't think I can upgrade from 3.0.1 to 3.1.5 as a result until I upgrade my ESCs to BLHeli and enable Oneshot. Maybe consider removing that restriction?﻿

Craft is a Eachine Racer 250, with Eachine F3 6DOF flight controller, Betaflight 3.0.1, 12A SimonK ESCs, and using the PWM ESC protocol.  The reason I'm NOT using Oncshot is of course because of the SimonK firmware on the ESCs.  I know they can be upgraded to BLHeli, but that's not the point.

borisbstyle, 2017-02-11:
2khz or 500hz should nothing have to do with motors not spinning up enough on take off.

Can you collect a blackbox log of that flight.

Btw having pid loop running at 2khz and motor protocol pwm means that pid loop runs 4x faster than esc can actually be updated

borisbstyle, 2017-02-11:
Also post your "diff" output and try to recalibrate escs maybe

Turismo05, 2017-02-11:
I would updated your esc's to the latest firmware to benefit from oneshot/multishot to take advantage of the awesomeness of these latest betaflight releases, is that not the whole point? faster loop times and esc update. I have and theres no going back!

spatzengr, 2017-02-12:
Please see below for link to blockbox at 0.5KHz, 0.57KHz, 1.0KHz and 2.0KHz.

[20170212-Racer250 BlackBox at differient PID freq.zip](https://github.com/betaflight/betaflight/files/769426/20170212-Racer250.BlackBox.at.differient.PID.freq.zip)

--------------------------------------------------
Diff output
--------------------------------------------------

# version
# BetaFlight/SPRACINGF3 3.0.1 Oct 18 2016 / 10:35:52 (48b7b4f)

# name

# mixer

# servo

# servo mix

# feature
feature -VBAT

# beeper

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100
aux 1 27 1 900 1400
aux 2 1 1 1700 2000
aux 3 2 1 1150 1450

# adjrange

# rxrange

# rxfail
rxfail 3 h
rxfail 5 s 1850

# master
set mid_rc = 1503
set min_check = 1030
set min_throttle = 1060
set motor_pwm_protocol = OFF
set gyro_sync_denom = 2
set deadband = 3
set yaw_deadband = 3
set failsafe_delay = 5
set rx_min_usec = 980
set rx_max_usec = 2020
set acc_trim_pitch = -30
set acc_trim_roll = 14
set pid_process_denom = 4

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

set rc_rate_yaw = 125
set rc_expo = 10
set thr_expo = 80

#

spatzengr, 2017-02-12:
It flies at all the frequencies except at 0.5KHz (as you can see in logs).

spatzengr, 2017-02-12:
Interesting.  I flashed 3.1.5 and in that release it does not restrict the PID loop to 0.5KHz when PWM is selected for ESC protocol.  If i set the PID loop to 2KHz in 3.1.5 it flies.  If i set to 0.5KHz, it doesn't even respond to throttle input from the transmitter.

borisbstyle, 2017-02-12:
All your logs appear to be made with 3.0.1. That doesnt help

spatzengr, 2017-02-13:
I can't show the issued at 0.5KHz PID loop in BF 3.1.5 as when i set the PID loop in 3.1.5 to 0.5KHz my copter will arm, but does not accept throttle input commands from the transmitter (can't throttle up).  With that, it does work for 0.57KHz+ PID loop frequencies - which i can now set as the PID loop restriction was removed that was in 3.1.0 - but it appears there is an issue in 3.0.1 and 3.1.5 when setting the PID loop frequency to 0.5KHz for some reason.  
 
Attached are backbox from 3.1.5.  Also below is my diff in 3.1.5.
[20170212-Racer250 BF 3.1.5 BlackBox at differient PID freq.zip.zip](https://github.com/betaflight/betaflight/files/769940/20170212-Racer250.BF.3.1.5.BlackBox.at.differient.PID.freq.zip.zip)

# DIFF
# Betaflight / SPRACINGF3 3.1.5 Feb  7 2017 / 22:30:44 (4646f9d)

feature -VBAT
feature -TELEMETRY
feature -CURRENT_METER
feature -TRANSPONDER
aux 0 0 0 1800 2100
aux 1 27 1 900 1400
aux 2 1 1 1500 2100
aux 3 2 1 1150 1450
aux 4 25 1 1500 1700
rxfail 3 h
rxfail 5 s 1850
set mid_rc = 1503
set min_throttle = 1060
set motor_pwm_protocol = OFF
set gyro_sync_denom = 2
set deadband = 3
set yaw_deadband = 3
set failsafe_delay = 5
set rx_min_usec = 980
set rx_max_usec = 2020
set pid_process_denom = 1
profile 0

rateprofile 0
rateprofile 0

set rc_rate_yaw = 125
set rc_expo = 10
set thr_expo = 80

#

Anisotropic, 2017-02-15:
@spatzengr do you use PWM receiver?

spatzengr, 2017-02-15:
My DSMX receiver uses PPM as the protocol to communicate to the FC.  Works fine when the PID loop frequencies in BF 3.1.5 is >0.5KHz.  However, if set to 0.5KHz it won't even respond to transmitter throttle commands.  

See https://www.youtube.com/watch?v=56xA_EG08VE

borisbstyle, 2017-02-20:
Issue is actually that your filters, which are actually default are trying to filter frequencies over nyquist frequencies :D.
Set your filter frequencies to be below nyquist frequency and it will work!

There is a guard for that added anyway in 3.1.6
https://github.com/betaflight/betaflight/pull/2367/commits/84105629ea945d2dd2eb4fbb215af74c482b0dee

closed 2017-02-20 by borisbstyle