betaflight/betaflight #2383: Support of external OSD via MSP gone in 3.1.x?

KritzelKratzel, 2017-02-12:
Is there no longer support of external OSD (like MinimOSD) in BetaFlight 3.1.x? 

I mean OSD that use the old MultiWii code with MSP_STATUS, MSP_ANALOG, etc. messages. Up to 3.0.1 this worked flawlessly on my FC (Flip32, UART2 -> MSP with 115_200 Bd)

blckmn, 2017-02-12:
Should still work no issues. Make sure MSP is turned on for the port it's connected to in the ports tab.

KritzelKratzel, 2017-02-12:
No, doesn't work. We have evidence with three different flight controls now (Flip32 (aka NAZE), SPRF3 and BlueJayF4). MSP was always turned on in the port tab with the right Baud setting. 

Original problem report: https://www.rcgroups.com/forums/showpost.php?p=36853802&postcount=237

Question: has the internal structure of the MSP message changed somehow in BF 3.1 or is it still compliant with the MultiWii structure description (http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol)?

mikeller, 2017-02-12:
There were some changes to some MSP commands, but all of these changes were of an additive nature, so they shouldn't break OSD if it's implemented properly. You'll probably have to enable debugging on your OSD to check where it's breaking.

jflyper, 2017-02-12:
It works.

OMNIBUS F4 + BF 3.1.5, Arduino Nano based MinimOSD compat + MWOSD R1.6.8.0+
NAZE (FLIP32AIO v1.02 + BF 3.1.5, MicroOSD v2.5 + MWOSD R1.6.8.0

Strange that you can't make FLIP32 work.

KritzelKratzel, 2017-02-12:
OK, I'll check it with my scope/port sniffer tomorrow and report back.

jflyper, 2017-02-12:
It may be a new MSP_DISPLAYPORT message that is injected at the boot time.

But funny, it has been there since 3.0...

@KritzelKratzel Waiting for your report.

jabacz, 2017-02-12:
I can confirm the issue: Nerdcam3D + BlueJayF4; Nerdcam3D + AirbotF4; Nerdcam3D + SPRF3
All the same issue: With BF 2.x and 3.0.1 it works well. After upgrade to 3.1 no data. Reverting to 3.0.1 fixes the problem.

mikeller, 2017-02-12:
@jabacz: Sounds like the Nerdcam3D's OSD is struggling with the changes to MSP that were introduced with 3.1. Having a debug log of the Nerdcam3D might help us see if something can be changed to make it work again. Or else themissinggear should update the firmware of the Nerdcam3D to work with Betaflight 3.1. Ideally the 3.0 -> 3.1 changes shouldn't be a problem for OSDs, but some of them are probably implemented in a way that can not cope with additive changes.

KritzelKratzel, 2017-02-13:
I finally found out what is going on. As of Betaflight 3.1.0-RC1 the MSP_STATUS message has an extended payload length. Following the MSP protocol available at (http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol) the payload of this message is exactly 11 bytes. 

Betaflight 3.0.1 (https://github.com/betaflight/betaflight/blob/v3.0.1/src/main/io/serial_msp.c#L718) adheres to this standard. The payload is:
```
cycleTime (2 bytes)
i2c_error_count (2 bytes)
sensors (2 bytes)
FlightModeFlags (4 bytes)
profile index (1 byte)
```
which makes in total 11 bytes.

In contrast, all Betaflight versions 3.1.x have an extended payload:
```
cycleTime (2 bytes)
i2c_error_count (2 bytes)
sensors (2 bytes)
FlightModeFlags (4 bytes)
profile index (1 byte)
averageSystemLoadPercent (2 bytes)
gyro cycle time (2 bytes and apparently not implemented)
```
which makes 15 bytes. The following two code lines break my current MSP implementation resulting in an invisible OSD:
```
sbufWriteU16(dst, constrain(averageSystemLoadPercent, 0, 100));
sbufWriteU16(dst, 0); // gyro cycle time

```
See https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_msp.c#L648. This extended status message was introduced with 3.1.0-RC1 (https://github.com/betaflight/betaflight/blob/v3.1.0-RC1/src/main/fc/fc_msp.c#L650). 

So far the analysis. I would like to raise two questions to the core developers ( @jflyper @mikeller ):

1. What was the motivation for extending the MSP_STATUS message?

mikeller, 2017-02-13:
Hi @KritzelKratzel.

Yes, sorry, there was this change, I'd forgotten about it totally.

Ironically, this change was made in order to make Betaflight's MSP work with another OSD implementation by making the format of MSP_STATUS identical to cleanflight's implementation, as described in #1735.

In general, handling of MSP messages should be done in a way that allows the messages to be extensible. That means, have your implementation parse the fields it knows about from the start of every message, and ignore any additional data after the last field you know about. The convention for new data to be added to MSP is to append the new fields at the end of the existing message, and not ever change the existing fields. This way, we can add new features to MSP without the risk of breaking existing clients.

If your OSD implementation does not yet conform to this convention, I would recommend you change it so it does, as not doing so will probably make it break on future changes as well.

KritzelKratzel, 2017-02-13:
I hate to write this, but I have to acknowledge your point. It's a pity that the message length in the original MSP description is not marked as potentially variable - or I have missed that upon implementing MSP in my camera firmware. 

I'll update my camera firmware to cope with variable message lengths of legacy MSP messages. Issue can be closed.

jflyper, 2017-02-14:
@KritzelKratzel BTW, why not implement the MSP_DISPLAYPORT handler in Nerdcam? It will give your Nerdcam an ability to run Betaflight CMS.

mikeller, 2017-02-14:
Totally agreed with @jflyper's suggestion. Also, engineering samples of your cam would be nice. :-P

closed 2017-02-14 by mikeller