betaflight/betaflight #2386: Sparky 2.0 Accelerometer

Tomcatt, 2017-02-12:
Hello BF.  I've updated my Sparky 2.0 and I've notice that no matter what I set my acc_harware setting to, I can't enable the accelerometer.  Is this a bug on the configurator part or what?  Don't want to take it up without it.

The switch is turned on under the configuration button.

mikeller, 2017-02-12:
This does not sound like a configurator issue. If it isn't highlighted in the configurator it's most likely not recognised. You can check this by running `status` in CLI. If it's not listed there, it's not recognised.

Somebody with a Sparky 2.0 board should verify / fix this.

Tomcatt, 2017-02-12:
[image]

In the past AUTO used to pull another driver but it'll still work.  I'll test it out but I'm not a coder.

mikeller, 2017-02-12:
As I suspected, it's not recognised. Can you post the same for older firmware that recognises the ACC, so we know what hardware is actually present?

Tomcatt, 2017-02-12:
I don't remember where I found the history of the SPARKY in BF (sorry, I didn't keep copies).  If anyone has a link, I'll test it.

mikeller, 2017-02-12:
You can select and flash an old version of Betaflight right in the configurator, on the 'firmware flasher' tab.

Tomcatt, 2017-02-12:
3.0.1 works

mikeller, 2017-02-12:
@Tomcatt: Can you post the output of `status` in CLI for 3.0.1 please?

Tomcatt, 2017-02-12:
[image]

v3.0.1

Ozzy33, 2017-02-13:
If you looks back 2 days ago, I posted the same issue. (#2376)

"I tried acc_hardware = MPU9250 (instead of AUTO), same.

A little more testing reveals that in 3.0.1 ACC is working, in 3.1.0, 3.1.3 and 3.1.5 it is not working."

mstrinzha, 2017-02-20:
Fix is available at https://github.com/betaflight/betaflight/pull/2445

blckmn, 2017-02-21:
@mstrinzha that fix will break other targets. Better fix is to remove 6500 driver from sparky, and explicitly use 9250.

Tomcatt, 2017-02-27:
Good news, I compiled the current github release and my acc_hardware is working again with v3.2.0.  I can set it to MPU9250 again.  However I can't give you a full answer yet; making some changes on the quad so as soon as I can, I'll take it up for you.

Tomcatt, 2017-03-11:
Thanks guys for fixing this.

closed 2017-03-11 by Tomcatt