betaflight/betaflight #2388: BlueJayF4 (BG clone) won't arm - think I've tried everything obvious...

RipperDrone, 2017-02-12:
- switch-arms ok when connected to battery AND USB
- does not arm when connected to battery only
- FCs red LED shows slow blink when connected to battery only (does this say gyro cal cannot be completed?)
- tried all of the following:
  - calibrated acc again
  - increased moron_threshold up to 100, 200 due to noisier gyro 20608
  - ran gyro/PID @ 4/2, 8/4, 8/8, CPU load is single digit
  - plugged off all current intense consumers like cam & VTX
  - plugged in battery first, then USB: Strangely, on the BFC start screen, the model would only move in slow motion when turning the copter around (gyro buffer?)

What else can keep the FC from arming? Or the gyro from calibrating? All sensor readings for gyro and acc look ok in the sensor view...

Any help is appreciated.

[Insect 130 Betaflight Dump Latest.txt](https://github.com/betaflight/betaflight/files/769727/Insect.130.Betaflight.Dump.Latest.txt)

mikeller, 2017-02-12:
> plugged in battery first, then USB: Strangely, on the BFC start screen, the model would only move in slow motion when turning the copter around (gyro buffer?)

This can be an indication of ACC calibration not yet being completed. You can debug this by setting `set debug_mode = gyro`, and then watching debug[3] in the sensors output. This shows the current noise level until the ACC calibration is complete (i.e. noise level below `moron_threshold`), and after that it stays constant.

ledvinap, 2017-02-13:
@mikeller : GYRO, not ACC. 

Maybe gyro is too noisy when powered from battery, increasing `moron_threshold` may mask it ..

borisbstyle, 2017-02-13:
@ledvinap gyros are very noisy when all filtering is disabled. The newer gyros have the config option for 32khz sampling and this is where the noise is much higher.
It may indeed be noise from power.

ledvinap, 2017-02-13:
@borisbstyle : Interesting thing is that USB powered gyro passes the test.. I would expect minimal difference on well-designed board

borisbstyle, 2017-02-13:
I believe the real bluejay f4 is externally powered. It uses the external power regulator. But this is a clone board. It may be nice if @RipperDrone could confirm that and post the detaills about his external power regulator?

RipperDrone, 2017-02-13:
Wow, so much supporting attitude - thanks, people!

Here is an update on further analysis: set debug_mode=gyro reading after plugging in battery 1st, USB 2nd:

[image]

RipperDrone, 2017-02-13:
This looks like it starts high @ > 100, then calms down (spike is from moving the FC around to check reaction). Will try moron_threshold of 100 again and wait longer for the gyro to settle...

RipperDrone, 2017-02-13:
I have the 32kHz gyro sampling OFF at all times - just to confirm...

RipperDrone, 2017-02-13:
Ok, tried several times now with moron_threshold set to 100. It takes 1-3min for the gyro to settle, then it beeps 3x and arms afterwards.

It seems like the gyro noise is huge after plugging the battery in and then takes "forever" to settle. What can it be - do I need to apply any non default filters to make sure noise levels come down more quickly?

RipperDrone, 2017-02-13:
@borisbstyle  @ledvinap @mikeller 

Since I do not exactly know the external power regulator components, here is a link to high-res photos of the FC, as well as circuit diagrams. Hope this helps to diagnose further:

https://www.unmannedtechshop.co.uk/streak-f4-v3-168mhz-flight-controller-with-osd/

RipperDrone, 2017-02-13:
You won't believe it - issue is resolved. That's the exciting and at the same time frustrating effect of this hobby: You always think you learnt something, but the next surprise is around the corner:

I had a battery strap running under the FC, and it seems it juuuust touched one of the ESC cables (which I routed under the FC) slightly. Somehow this touching cable transferred a little restraint on the FC, so the gyro obviously kept reading signals and did not come to rest. 

I discovered the effect when dismantling the battery strap completely, just by chance. 3 beeps, armed. Put it back on: no beeps, no arm.

Now I bent the ESC cable out of the way, not touching the strap anymore. And gotcha: Arms anytime :-)

Wow, the gyro must be sooo sensitive!

Thanks for all your great inputs - happy for the moment (until next challenge comes...) ;-)

mikeller, 2017-02-13:
:-)

Sometimes it's unexpected things that go wrong.

RipperDrone, 2017-02-13:
Looking at the noise / sensitivity of that gyro, it seems this guy is never gonna fly on my Insect 130: Even lowpass filtering does not yield a stable PID control loop, after arming the poor thing quickly ends up in twitch spasm - seems it is close to impossible to mount this AIO FC in a way to keep noise / gyro sensitivity under control :-(

RipperDrone, 2017-02-13:
As said before: Resolving one issue just takes you to the next :-)

closed 2017-02-13 by RipperDrone