betaflight/betaflight #2389: OSD: Some improvements

maxsymba, 2017-02-13:
IMHO, osd-artificial-horizon at present time works in the wrong direction and have to be inverted by roll: when we roll to left - it must roll right. And vice versa.

P.S. Whether it is possible to add an feature for choosing an alternative crosshair, like the #define ALT_CENTER in MW_OSD?

jflyper, 2017-02-13:
I'm going to fix that.

crosshair is todo.

maxsymba, 2017-02-13:
Okay, thanks for the reply.

DanNixon, 2017-02-13:
@jflyper WHat is your plan with crosshairs?

As far as I'm concerned users can have whatever they want by just uploading their own font.

jflyper, 2017-02-13:
@DanNixon Crosshair feature must be implemented first...?
TBH, it's relatively low in my list.

DanNixon, 2017-02-13:
We already have a single character crosshair.

Is what you mean something larger?

maxsymba, 2017-02-13:
MWOSD have an alternative mode for crosshair - "rectangle by corners" instead of "cross with point at center",
All necessary symbols are present in default font.

VvIRONWOLFvV, 2017-04-12:
what do i do to fix horizon rolling wrong way? kinda new