betaflight/betaflight #2392: [Discussion] VTX consolidation phase 2 : RC commands

jflyper, 2017-02-13:
The VTX consolidation will go on to phase 2 for v3.2, and will adopt the SINGULARITY-style RC commands. See [SINGULARITY manual](http://singularity.impulserc.com/assets/files/manual.pdf) for details of VTX control related features.

Points of considerations are

- Minimum changes to exisiting SINGULARITY users; no significant CLI variable/command changes.
- Co-existence with CMS and vtxCommon based controls.

Here is my plan.

- CMS for VTX will have mode switching menu entry or mechanism that can modify `vtx_mode` variable.

For each mode, CMS behaves as follow.

- `vtx_mode = 0` (Band/Channel mode, using variables `vtx_band` and `vtx_channel`)
This mode allows RC stick commands to adjust band and channel.
Can co-exist with CMS, too; CMS for SmartAudio and Tramp already support Band/Channel mode.
Changes made with RC stick commands are reflected to CMS transparently, and vice-versa.

- `vtx_mode = 1` (Frequency mode, using variable `vtx_mhz`)
This mode has no RC control.
Can co-exist with CMS, too: CMS for SmartAudio does have support for user specified frequency mode. CMS for tramp must be expanded to implement this mode (minor addition to CMS menu structure).
Changes made in CMS will be reflected to vtx_mhz variable transparently.

- `vtx_mode = 2` (AUX switch selecting frequency pre-defined by `vtx` command).
This is the most problematic case in terms of inter-operability with CMS, because CMS can't change to frequency/band/channel that conflict with what is selected by AUX switches.
I propose CMS to disable band/channel/frequency changes under this mode.
RF power may be controllable.

Thoughts?

I especially like to hear from @sblakemore.

jflyper, 2017-02-13:
Another aspect to consider is conflict between `vtx_mode` and device-stored states.

Certain device, notably TBS SmartAudio stores frequency selection mode internally, and if `vtx_mode` is modified while the device is offline, it may conflict with what is stored on the device. Precedence of mode selection must be determined.

raphaelcoeffic, 2017-02-13:
Corresponding enhancements must be made available by MSP commands.

jflyper, 2017-02-13:
> Corresponding enhancements must be made available by MSP commands.

Correct. I think

1. Mode selection among Band/Channel, Direct frequency, RC channel (AUX)
2. Direct frequency selection API

are required.

On the other hand, it is interesting to see how "RC channel (AUX)" mode can be implemented. This will be outside the S.PORT/MSP scheme ...

raphaelcoeffic, 2017-02-13:
By the way, I think we need to be able to setup some "race" mode, and a switch to exit low power mode. This has been required by several racers.
@boris.b. Is that something you need as well?

jflyper, 2017-02-13:
@raphaelcoeffic @borisbstyle 
We have to officially define what "Betaflight VTX race mode" is.
The SmartAudio CMS already use RACE and FREESTYLE as two different modes of operation (this is called operational model).

https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#operational-models
> Operational Models
>
>In Betaflight, a SmartAudio device operates in one of two operational models:
>
>Race
>
>This is a model that gives minimum interferance to other pilots. A SmartAudio device powers up in pit mode, and remain in pit mode until transmission is commenced.

jflyper, 2017-02-13:
BTW, this is a depicted version of "My plan".
[image]

The CMS is separated from the `vtx_common` abstraction, because there are lots of device-dependent handling (labelled `cms`). Ideally speaking, the `cms` and `CMS` should also be generic that uses `vtx_common` abstraction, but we don't have enough experience yet to tell how to do it or it if is possible.

sblakemore, 2017-02-16:
I'm not really up to date with the current state of this, but it all sounds sensible. However there is another architecture which we have right now in the Helix that might be worth considering when making decisions.

We have a Minim OSD compatible board connected over UART and the OSD controls the RF module via SPI. Currently this is done via a fork of MW_OSD but should be officially supported in the next release.

I think from when I last looked external OSDs were being supported by BF using a canvas type abstraction where the OSD just puts characters where it is told. The problem with that is in our case the OSD also needs to know at least stick positions and armed status so it can control VTX power/channel.

I know this throws a spanner in the works and I'm not sure if anyone else will ever follow us with an architecture like this, but wanted to mention it.

jflyper, 2017-02-16:
@sblakemore I'm with MWOSD pretty much and have seen your code for helix as there it is going to be merged very soon (I think it's very well written).

And yes, I will probably add another extension for controlling devices (VTX more specifically), so that VTX on MWOSD can be controlled from CMS and VTXRC (Will probably be a bidir protocol because FC-side wants to know the current status of the VTX; which is another challenge in the current MSP framework.)

raphaelcoeffic, 2017-02-16:
As a side-node, from what we could see of the Vortex150, I think it is done the same there (OSD controlling VTX). They use MSP as well to query changes that have been requested by the Spektrum radio.

hydra, 2017-02-26:
some thoughts ideas, take your pick.
* need list-of-channels for a switch to use (2pos/3pos/rotary) to allow easy frequency changing.
 - e.g. 2pos switch, switch between a Boscam and a Raceband frequency depending on the goggles they're using on the day.
- 3 pos switch to switch between 3 different band/freqency combos.
- rotary switch to swtich between all channels on a band.
- a rotary switch and a momentary 2 position switch, one for channel selection, one for band cycle.
- two momentary switches, one for channel one for band cycling.
- a 3 position switch, up for channel cycling, down for channel cycling.
* need list-of-power settings for a switch to use (2pos/3pos/rotary) to allow easy race environment usage.
 - e.g. some vtx modules have LOW/HIGH, some have OFF/LOW/HIGH, some have OFF/PIT/25/200/600 but the user might want a 2pos switch to just OFF/ON, or a 3pos switch to do either OFF/LOW/HIGH or 25/200/600 and possibly another switch for OFF/ON (2 switches would require extra code to ensure they don't conflict)
* team-racers need to turn the VTX to OFF or LOW POWER mode via a switch if they crash in the field and another team member needs to race on the same frequency.

hydra, 2017-02-26:
a possible pseudo-config on how some of this could work would be as follows:

```
vtxfunc 0 OFF
vtxfunc 1 LOW
vtxfunc 2 HIGH

// AUX 1 = 3pos switch
adjrange AUX1 1000..1250 vtxfunc 0
adjrange AUX1 1250..1750 vtxfunc 1
adjrange AUX1 1750..2000 vtxfunc 2

// AUX 2 = 2pos switch
adjrange AUX2 1000..1500 vtxfunc 0
adjrange AUX2 1500..2000 vtxfunc 2

----

vtxfunc 0 CYCLE_BAND
vtxfunc 1 CYCLE_CHANNEL

// AUX 1 = 3pos switch
adjrange AUX1 1000..1250 vtxfunc 0
adjrange AUX1 1750..2000 vtxfunc 1

---

vtxfunc 0 SET BAND A
vtxfunc 1 SET BAND R
vtxfunc 2 SET CHANNEL 1
vtxfunc 3 SET CHANNEL 8

// AUX 1 = 2pos switch to switch between A1 and R8
adjrange AUX1 1000..1250 vtxfunc 0
adjrange AUX1 1750..2000 vtxfunc 1

adjrange AUX1 1000..1250 vtxfunc 2
adjrange AUX1 1750..2000 vtxfunc 3

---

vtxfunc 0 MAP TO BAND
vtxfunc 1 MAP TO CHANNEL

// AUX1 and 2 are rotary switches (e.g pots on the taranis)
adjrange AUX1 1000..2000 vtxfunc 0
adjrange AUX2 1000..2000 vtxfunc 1

```

the idea is that you define a VTX function, and then map an adjustment range to it.

jflyper, 2017-03-24:
@sblakemore What is the default behavior of the SINGULARITY-style RC commands in mode 2 (AUX switch) when none of the activation rules match?

hydra, 2017-04-15:
note, i deleted vtx_mode as part of https://github.com/cleanflight/cleanflight/pull/2721 - The VTX API didn't have support for setting a particular frequency.  The code in the driver files remains however.